rusefi/rusefi #2028: basic hw in the loop - pwm self validation

rusefillc, 2020-12-06:
we already have hw pins on test discovery

we already have logic capture code we just need to revive if broken

we have all pieces to self check eth gppwm waste etc duth cycle at lease

this might push us to apply more settings on the fly more better which I would love to see

rusefillc, 2020-12-06:
@mck1117  says
you can probably just hw sniff the same pin that we output to
since IDR reads the actual state of the pin even in output mode

rusefillc, 2020-12-09:
set_trigger_simulator_pin 0 none
set_trigger_simulator_pin 1 none
set_idle_pin PD1

set_ignition_pin 1 none
set_ignition_pin 2 none
set_ignition_pin 3 none
set_trigger_input_pin 0 PC7

set_logic_input_pin 0 pc6

rusefillc, 2020-12-13:
related to #2069

rusefillc, 2020-12-15:
#2081

rusefillc, 2020-12-15:
@shadowm60 can you move this ticket forward?

you have commands above ^^^
you have your new debug mode
now just need to invoke those commands from a test and validate output

See https://github.com/rusefi/rusefi/blob/master/java_console/autotest/src/com/rusefi/VssHardwareLoopTest.java as a very similar test

please download https://www.jetbrains.com/idea/download/ for java IDE https://www.jetbrains.com/idea/download/

mck1117, 2021-06-26:
@shadowm60 do we think this is done?

shadowm60, 2021-06-29:
I don't think so, it did not had time to talk with @rusefillc about this ticket, partially done, but i've got stuck with java, it was working, but not sure if we consider it done or is there something left to do.