rusefi/rusefi #2049: GDI injector deadtime to like flat 0.2ms

rusefillc, 2020-12-09:
B6 config

you should set your injector deadtime to like flat 0.2ms

no compensation for battery voltage, boost voltage on gdi board is constant

mck1117, 2020-12-09:
Why does this need an issue? Just set the table values to all 200us or whatever.

rusefillc, 2020-12-09:
my process is engine config in the code. I do not manually deal with tables since too many devices vehicles and context switches. GIT repo is the storage for vehicle configurations for me.

rusefillc, 2020-12-11:
Also need to hard-code:

```
1214 cc/min
"High"
"Sensed Rail Pressure"
10000 kPa
```

rusefillc, 2021-01-22:
this ticket is about default settings in https://github.com/rusefi/rusefi/blob/master/firmware/config/engines/vw_b6.cpp

closed 2021-01-24 by rusefillc