rusefi/rusefi #2058: FSIO better handling of invalid function usage

rusefillc, 2020-12-12:
@shadowm60 has attempted to use "clt" instead of "coolant" and error message was not helpful

we shall improve handling of invalid functions. Maybe FATAL error in case of invalid function? maybe not.

Maybe nice clear warning message at least in rusEFI console?

rusefillc, 2020-12-12:
@mck1117 says "100% fatal in case of invalid function" which makes implementation really easy

@shadowm60 can you try to implement critical error in case of invalid function in fsio?

shadowm60, 2020-12-12:
Yes i can try it!

mck1117, 2020-12-12:
probably near `fsio_core.cpp` line 425

shadowm60, 2020-12-12:
There is where we can signal, but i did not noticed that warning, so i debug a bit, to see how did clt string did not generated the warning, or since it was a warning, it just went on the log and i did not noticed.

rusefillc, 2020-12-12:
So we have proof that warnings do not work here, we either need to make warnings way smarter or just promote this to a harder-to-ignore critical

mck1117, 2020-12-12:
just make it a critical - there is no world in which it fires accidentally and isn't an actual error

rusefillc, 2020-12-12:
for now let's make it critical

there is world where people make a type and notice the typo right away. ideally we might throw fatal 10 seconds after some sort of visible warning? but I am not sure if we have human-readable visible warnings in TS?

mck1117, 2021-11-29:
goodbye fsio

closed 2021-11-29 by mck1117