rusefi/rusefi #2065: MRE issue with tle8888

shadowm60, 2020-12-12:
with sw 20201008 in my case GP3 was used as output (GPPWM).
with SW since TLE driver update i had issues but with latest nightly i had time to investigate.
I have excluded GPPWM and FSIO from the scheme since i did the following:
assign GP3 as ignition1, and kept ignition2 for ignition, then started simualtor and logged pin activity:
This is the output:
https://i.imgur.com/z0nnFd3.png
reverted to the old SW to ensure that hw is ok and TS config as well and the putput was ok:
https://i.imgur.com/SgirIy0.png

With a short debug session i noticed that we hit periodic  reinit request from tle8888_spi_validate since we keep getting reg 0 as a response.

rusefillc, 2020-12-13:
is that F4 or F7 MRE @shadowm60 ?

rusefillc, 2020-12-13:
i think QC build reproduces the problem on F4 - GP1 & GP2, LS1 & LS2 do blink but GP3 does NOT blink!

shadowm60, 2020-12-13:
I have mre 0.3 it is f7

dron0gus, 2020-12-13:
@shadowm60 
Can you post output of "tle8888" and "pins" commands when you hit this issue?
I'm trying to reproduce this on MRE 0.5.0 with F4 with no luck yet.
Do you see this issue with engine_type 31 and self stimulation?

shadowm60, 2020-12-13:
@dron0gus  How would you like the outputs? console? or how should i prepare it?

shadowm60, 2020-12-13:
Changed engine config to 31
https://i.imgur.com/NJboKp4.png 
Ch1 is gp3 and nothing on that line, the ch2 is ignition, just to see if the ecu is running with self stim.

020-12-13_18_09_57_205: EngineState: SPI clock on PB3
2020-12-13_18_09_57_207: EngineState: SPI master out on PB5
2020-12-13_18_09_57_208: EngineState: SPI master in  on PB4
2020-12-13_18_09_57_210: EngineState: SPI clock on PC10
2020-12-13_18_09_57_211: EngineState: SPI master out on PC12
2020-12-13_18_09_57_212: EngineState: SPI master in  on PC11
2020-12-13_18_09_57_214: EngineState: tle8888 CS on PD5
2020-12-13_18_09_57_215: EngineState: SD CS on PB9
2020-12-13_18_09_57_216: EngineState: Injector 1 on ext:TLE8888.OUT6
2020-12-13_18_09_57_217: EngineState: Injector 2 on ext:TLE8888.OUT5
2020-12-13_18_09_57_218: EngineState: Injector 3 on ext:TLE8888.INJ4
2020-12-13_18_09_57_219: EngineState: Injector 4 on ext:TLE8888.INJ3
2020-12-13_18_09_57_220: EngineState: Injector 5 on ext:TLE8888.INJ2
2020-12-13_18_09_57_221: EngineState: Injector 6 on ext:TLE8888.INJ1
2020-12-13_18_09_57_222: EngineState: Injector 7 on PA4
2020-12-13_18_09_57_223: EngineState: Injector 8 on PB1
2020-12-13_18_09_57_225: EngineState: Injector 9 on PB0
2020-12-13_18_09_57_226: EngineState: Injector 10 on PC4
2020-12-13_18_09_57_226: EngineState: Injector 11 on ext:TLE8888.OUT13
2020-12-13_18_09_57_227: EngineState: Injector 12 on ext:TLE8888.OUT10
2020-12-13_18_09_57_228: EngineState: Coil 1 on PD6
2020-12-13_18_09_57_236: EngineState: Coil 2 on PD7
2020-12-13_18_09_57_237: EngineState: Coil 3 on PD1
2020-12-13_18_09_57_238: EngineState: Coil 4 on PD2
2020-12-13_18_09_57_239: EngineState: Coil 5 on PD3
2020-12-13_18_09_57_240: EngineState: Coil 6 on PD4
2020-12-13_18_09_57_241: EngineState: Coil 7 on ext:TLE8888.OUT11
2020-12-13_18_09_57_242: EngineState: Coil 8 on ext:TLE8888.OUT12
2020-12-13_18_09_57_243: EngineState: Coil 9 on ext:TLE8888.OUT21
2020-12-13_18_09_57_244: EngineState: Coil 10 on ext:TLE8888.OUT22
2020-12-13_18_09_57_245: EngineState: Coil 11 on ext:TLE8888.OUT23
2020-12-13_18_09_57_246: EngineState: Coil 12 on ext:TLE8888.OUT24

I have debug channel TLE8888 set, but in console i do not have anything extra. only some warnings about CAM signal.

Did you measured it with oscope?

@rusefillc confirmed this issue on f4, but i do not know the hw wariant. On 0.3 to 05 there is no difference in tle spi and tle chip, or do we have different IC versions?

dron0gus, 2020-12-13:
Ok.
Reproduced on 0.4.8-2L board with TLE8888-2QK (no watchdog enabled by default).
Works just fine on 0.5.0 and 0.4.7-2L with TLE8888QK (watchdog enabled by default).
Some watchdog-related issue?

rusefillc, 2020-12-13:
WOW what the hell, how is this related to watchdog?!?!?!?!

errata time?

mck1117, 2020-12-13:
> errata time?

but it worked on the same hardware on old firmware

shadowm60, 2020-12-14:
Different versions of tle manual used in implementation?

dron0gus, 2020-12-14:
There is only one datasheet for TLE8888. And one app-note for WD-enabled version of chip that defines WD register settings.
This is my fault I was testing last code on HW with watchdog enabled chip only...

dron0gus, 2020-12-14:
Version with WD disabled. Driver keeps feeding WD. WD never becomes happy. Driver see this and never calls tle8888_update_output() which should update all non-direct outputs.
Attempt to setup WD with same settings as TLE8888-2QK also does not make non-WD chip happy.
Will continue investigating.

Workaround: https://github.com/rusefi/rusefi/pull/2079

shadowm60, 2020-12-14:
I will try this fix today and report back.

shadowm60, 2020-12-14:
Is there a way to read it from SW if we have an ic with WD enabled or not? then we could treat it differently

dron0gus, 2020-12-15:
To determine what type of chip it is we can read settings registers. Ones that fixes on WD-enabled chip. Or try to modify and read back.
But I don't want to thread these two version differently. This may be a cause for new bugs. :)

dron0gus, 2021-02-04:
I think this can be closed?

closed 2021-02-04 by rusefillc