rusefi/rusefi #2069: logic_analyzer to report frequency and duty cycle into debug outputs

rusefillc, 2020-12-13:
We have logic_analyzer.cpp which I would like to use for self-hw-validation with a jumper

at the moment we have all the commands for java test framework to turn logic_analyzer.cpp on/off but we do not have efficient way for logic_analyzer.cpp to return data into console, text based protocol is probably not great for that

we need a new DBG_ debug_mode for logic_analyzer.cpp which would report frequency and duty cycle into debug outputs

shadowm60, 2020-12-14:
Just checked the sources, we init 4 channels,
We can report like this:
f1 -  ch1 high_time [us]
f2 - ch2 high_time [us]
f3 - ch3 high_time [us]
f4 - ch4 high_time [us]
f5 - ch1 freq in float [hz] ? or should reuse it for something else?
f6 - ch2 freq in float [hz] ? or should reuse it for something else?
f7 - ch3 freq in float [hz] ? or should reuse it for something else?
I1 - ch1 freq in Hz
I2 -ch2 freq in Hz
I3 - ch3 freq in Hz
I4 - ch4 freq in Hz

For freq in think it should be ok with the step of 1Hz, or do we need this to be more fine measurement?

rusefillc, 2020-12-14:
i would do
f1-f4 duty cycle in %

i1-i4 frequency as interest

high_time we can calculate from duty cycle and frequency

shadowm60, 2020-12-14:
OK, i just needed the information on how to report the values.

rusefillc, 2020-12-14:
add into outputs cycle, see where many are assigned regularly