rusefi/rusefi #2077: export more of trigger attributes into triggers.txt file

rusefillc, 2020-12-14:
At the moment most of trigger parameters are defined in C code and exported into triggers.txt for TriggerImage java code to render, but some stuff is hard coded in java

1) human-readable trigger name
2) isFirstCrankBased
3) isSecondCamBased

all data about trigger should be defined in one place and at the moment that one place is C trigger definition structure

We should remove all hard-coded stuff from java and hard-code anything missing in C

rusefillc, 2021-02-08:
```
; todo: code generator to hard-code all triggers not requiring VVT?	    
	    field = "CAM/VVT mode shaft #1",				vvtMode, {trigger_type != TT_TT_MAZDA_MIATA_NA}
```

that's another reason to flip trigger workflow

instead of C generating triggers.txt we shall have triggers.yaml generating C and java and .ini

rusefillc, 2021-02-08:
@chuckwagoncomputing how can I trick you into this issue?

chuckwagoncomputing, 2021-02-13:
> instead of C generating triggers.txt we shall have triggers.yaml generating C and java and .ini

Which .ini?

rusefillc, 2021-02-13:
.input
https://github.com/rusefi/rusefi/blob/master/firmware/tunerstudio/rusefi.input#L1808
and
https://github.com/rusefi/rusefi/blob/master/firmware/tunerstudio/rusefi.input#L1820

.txt
https://github.com/rusefi/rusefi/blob/master/firmware/integration/rusefi_config.txt#L653

chuckwagoncomputing, 2021-02-13:
So the trigger info should be taken out of those files, and the yaml should be passed to ConfigDefinition.
Do I understand?

rusefillc, 2021-02-13:
Also most of https://github.com/rusefi/rusefi/tree/master/firmware/controllers/trigger/decoders should be replaced by something generated by new java code based on triggers.yaml

And https://github.com/rusefi/rusefi/blob/master/java_console/ui/src/main/java/com/rusefi/TriggerImage.java#L50

rusefillc, 2021-02-13:
by the way firing order tool is somewhat similar just much simpler

rusefillc, 2021-04-29:
Depends on #2607

rusefillc, 2022-04-03:
Wow a sad discovery #4032

chuckwagoncomputing, 2023-02-04:
What is the current state of this?

rusefillc, 2023-02-04:
a lot of progress was done!

at least nice names are still in java and should be moved into firmware as EFI_UNIT_TEST field? Actually human-readable name could be the last thing to move out of java

```
    private static String getTriggerName(TriggerWheelInfo triggerName) {
        switch (findByOrdinal(triggerName.getId())) {
            case TT_FORD_ASPIRE:
                return "Ford Aspire";
            case TT_VVT_BOSCH_QUICK_START:
                return "Bosch Quick Start";
            case TT_MAZDA_MIATA_NA:
                return "Miata NA";
            case TT_MAZDA_MIATA_NB1:
                return "Miata NB";
            case TT_SUBARU_SVX:
                return "Subaru SVX";
            case TT_HONDA_K_CRANK_12_1:
                return "Honda K 1/12";
            case TT_SUBARU_7_6:
                return "Subaru 7/6";
            case TT_GM_24x:
```

rusefillc, 2026-02-14:
@chuckwagoncomputing I am out of context

Why does anything need to happen to initializeSkippedToothTrigger helped method? how is initializeSkippedToothTrigger related to TriggerImage?