rusefi/rusefi #2081: new digital input driver: software busy read

rusefillc, 2020-12-15:
ICU can only generate events on 8 pins, PAL can only generate events on 16 pins - that's just too little options

A 100% software input capture driver could be a thread reading from all specified pins in a constant loop, with lowest possible thread priority

shadowm60, 2020-12-15:
i think here we would need to define the frequency range we should be able to read, since usually frequencies are not that high, we can read pins in a while look (cyclic value would be predefined).

mck1117, 2020-12-15:
Yes, not an unreasonable idea.  Running on a super low priority thread will have (in theory) zero impact on anything else running already on the ECU.