rusefi/rusefi #2085: HW CI: debounce via start/stop

rusefillc, 2020-12-17:
we need a java HW CI test for debounce, using start/stop feature

0) set DBG_START debug mode
1) assign start button pin
2) feed some PWM into it
3) assert "Start count" in debug fileds

chuckwagoncomputing, 2021-01-09:
This would be done in FunctionalTest.java, right?

rusefillc, 2021-01-09:
yes https://github.com/rusefi/rusefi/blob/master/java_console/autotest/src/com/rusefi/FunctionalTest.java is a great option

chuckwagoncomputing, 2021-01-11:
You can assign this to me. I haven't done any HW CI stuff yet so it may take a bit for me to figure it out though.

rusefillc, 2021-01-11:
you need discovery and two "official" jumpers, also IDEA IDE really helps with java

chuckwagoncomputing, 2021-01-11:
Is there a wiki page or something that tells me how to hook up the jumpers?

rusefillc, 2021-01-11:
https://github.com/rusefi/rusefi/wiki/Dev-Quality-Control

chuckwagoncomputing, 2021-01-11:
How do I assign pins and do IO? Do I have to write a console command that does all the setup for the test, or can I do it from Java? I can't tell that any of the other HW CI tests do it.

rusefillc, 2021-01-12:
java invokes commands
firmware has to have all required commands available

java tells firmware what to do via commands

technically java has "set_int x y" command to set any setting to Y by offset X but that's less cool

rusefillc, 2021-01-12:
and we will use GPPWM as feed data for start/stop
or maybe use "bench test" as data source? "bench test" of something is at least specified number of events

one way or another, we need
- signal source
- signal consumer
- way for java to validate