rusefi/rusefi #2102: generate java enum from C enum

rusefillc, 2020-12-18:
at the moment in order to sync C and java enums we have the 
```
#define ET_DODGE_NEON_1995 2
#define ET_FORD_ASPIRE 3
#define ET_FORD_FIESTA 4
```
in .txt while both C and java is generated

it would be more efficient to generate some java based on C .h header without .txt involvment

rusefillc, 2020-12-18:
we already read C .h see gen_enum_to_string.sh so some building pieces are already there

rusefillc, 2021-02-13:
```
typedef enum {
	DEFAULT_FRANKENSO = ET_DEFAULT_FRANKENSO,
	/**
	 * 1995 Dodge Neon
	 * http://rusefi.com/forum/viewtopic.php?t=360
	 */
	DODGE_NEON_1995 = ET_DODGE_NEON_1995,
	/**
	 * 1996 1.3 Ford Aspire
	 * http://rusefi.com/forum/viewtopic.php?t=375
	 */
	FORD_ASPIRE_1996 = ET_FORD_ASPIRE,

	NISSAN_PRIMERA = 5,
	HONDA_ACCORD_CD = 6,

```

chuckwagoncomputing, 2021-02-22:
Just to be clear, we decided that we want to generate the Java enum from the C enum, and not have a .yaml. Is that correct?

rusefillc, 2021-02-22:
yes, not yaml as source but C

rusefillc, 2021-02-22:
but that leaves question of how to define human-readable TS literals

chuckwagoncomputing, 2021-02-22:
What about writing C from the info from rusefi_config.txt?

chuckwagoncomputing, 2021-02-22:
We are already writing rusefi_generated.h. Does that not contain all the information that C needs, just not in an enum?

rusefillc, 2021-02-22:
text has horrible by index management, it's a nightmare to support

@mck1117 @andreika-git ideas?

chuckwagoncomputing, 2021-02-22:
Where are the TS engine types currently? I see the trigger type names in rusefi_config.txt, but not engine type names.

rusefillc, 2021-02-22:
engine types not in TS, I was talking generally

output pins would be a case of scary enum drama for TS >> C mapping

chuckwagoncomputing, 2021-02-23:
So you are interested in reading more things from C and removing them from rusefi_config.txt, and want to follow the same pattern?

rusefillc, 2021-02-23:
there is clearly a problem with manual sync of entities but also no great solution at the moment

I am not happy with yaml as source of truth here, my preference is C as source of enums

but literals are needed, maybe extend java enum reader to parse

VAL/*name=Value*/=3,