rusefi/rusefi #2107: Gppwm is missing from output once you hit burn.

shadowm60, 2020-12-19:
With latest nightly build is the same. In my case mre 0.3 gp3 pwm will stop as soon as i hit burn.

I will test with other outputs and report back with the result.

mck1117, 2020-12-24:
@shadowm60 can you share a config that reproduces this?

shadowm60, 2020-12-26:
So my tune is on REO: https://rusefi.com/online/view.php?msq=301

It is GP3 out 3 that is acting up, i did not had time to test other outputs to see if it is only connected to GP3. My MRE 0.3 hat issues with GP3 since the tle update, now only this issue is remaining since i control my fan with this output and once i hit burn it would start with full power :(

shadowm60, 2020-12-27:
Ok, i did the following test, GP3 set as tacho output, enableing the self stim it was working at 1200 rpm, i have changed the RPM to 5000 burn, and tacho was dead, then connected tacho to GP6 and it was working fine even after burn, i noticed reinit of pin after burn, but at the end it was working, but i cannot say the same of GP3.

mck1117, 2020-12-30:
ok, I've successfully repro'd this bug now on a 0.4.8 board with TLE8888-2QK

mck1117, 2020-12-30:
this seems to be a continuation of #2065

mck1117, 2020-12-30:
@dron0gus I don't know enough about the tle8888 driver to fix this right now, so you'll have to pick this one up

mck1117, 2021-01-05:
@dron0gus ping

dron0gus, 2021-01-05:
I'm here. Back from holidays. Start looking at this.

dron0gus, 2021-01-06:
Looks like SPI communication is broken after flashing new settings.
"tle8888" command reports following:
```
2021-01-06_14_05_47_665: EngineState: register: data
2021-01-06_14_05_47_666: EngineState: 7F: FF
2021-01-06_14_05_47_667: EngineState: 7F: FF
2021-01-06_14_05_47_668: EngineState: 7F: FF
2021-01-06_14_05_47_668: EngineState: 7F: FF
2021-01-06_14_05_47_669: EngineState: 7F: FF
2021-01-06_14_05_47_669: EngineState: 7F: FF
2021-01-06_14_05_47_669: EngineState: 7F: FF
2021-01-06_14_05_47_670: EngineState: 7F: FF
2021-01-06_14_05_47_670: EngineState: 7F: FF
2021-01-06_14_05_47_671: EngineState: 7F: FF
2021-01-06_14_05_47_671: EngineState: 7F: FF
2021-01-06_14_05_47_671: EngineState: 7F: FF
2021-01-06_14_05_47_672: EngineState: 7F: FF
2021-01-06_14_05_47_677: EngineState: 7F: FF
2021-01-06_14_05_47_677: EngineState: 7F: FF
2021-01-06_14_05_47_678: EngineState: 7F: FF
2021-01-06_14_05_47_678: EngineState: 7F: FF
2021-01-06_14_05_47_689: EngineState: 7F: FF
2021-01-06_14_05_47_689: EngineState: 7F: FF
2021-01-06_14_05_47_690: EngineState: 7F: FF
2021-01-06_14_05_47_690: EngineState: 7F: FF
2021-01-06_14_05_47_693: EngineState: 7F: FF
2021-01-06_14_05_47_694: EngineState: 7F: FF
2021-01-06_14_05_47_695: EngineState: 7F: FF
2021-01-06_14_05_47_695: EngineState: 7F: FF
2021-01-06_14_05_47_705: EngineState: 7F: FF
2021-01-06_14_05_47_705: EngineState: 7F: FF
2021-01-06_14_05_47_705: EngineState: 7F: FF
2021-01-06_14_05_47_706: EngineState: 7F: FF
2021-01-06_14_05_47_706: EngineState: 7F: FF
2021-01-06_14_05_47_712: EngineState: 7F: FF
2021-01-06_14_05_47_713: EngineState: 7F: FF
2021-01-06_14_05_47_713: EngineState: 7F: FF
2021-01-06_14_05_47_714: EngineState: 7F: FF
2021-01-06_14_05_47_718: EngineState: 7F: FF
2021-01-06_14_05_47_718: EngineState: 7F: FF
2021-01-06_14_05_47_718: EngineState: 7F: FF
2021-01-06_14_05_47_722: EngineState: 7F: FF
2021-01-06_14_05_47_722: EngineState: 7F: FF
2021-01-06_14_05_47_722: EngineState: 7F: FF
2021-01-06_14_05_47_723: EngineState: 7F: FF
2021-01-06_14_05_47_723: EngineState: 7F: FF
2021-01-06_14_05_47_726: EngineState: 7F: FF
2021-01-06_14_05_47_726: EngineState: 7F: FF
2021-01-06_14_05_47_726: EngineState: 7F: FF
2021-01-06_14_05_47_727: EngineState: 7F: FF
2021-01-06_14_05_47_727: EngineState: 7F: FF
2021-01-06_14_05_47_731: EngineState: 7F: FF
2021-01-06_14_05_47_734: EngineState: 7F: FF
2021-01-06_14_05_47_734: EngineState: 7F: FF
2021-01-06_14_05_47_734: EngineState: 7F: FF
2021-01-06_14_05_47_734: EngineState: 7F: FF
2021-01-06_14_05_47_735: EngineState: 7F: FF
2021-01-06_14_05_47_737: EngineState: 7F: FF
2021-01-06_14_05_47_738: EngineState: 7F: FF
2021-01-06_14_05_47_738: EngineState: 7F: FF
2021-01-06_14_05_47_738: EngineState: 7F: FF
2021-01-06_14_05_47_739: EngineState: 7F: FF
2021-01-06_14_05_47_739: EngineState: 7F: FF
2021-01-06_14_05_47_740: EngineState: 7F: FF
2021-01-06_14_05_47_743: EngineState: 7F: FF
2021-01-06_14_05_47_743: EngineState: 7F: FF
2021-01-06_14_05_47_744: EngineState: 7F: FF
2021-01-06_14_05_47_744: EngineState: 7F: FF
2021-01-06_14_05_47_749: EngineState: 7F: FF
2021-01-06_14_05_47_749: EngineState: 7F: FF
2021-01-06_14_05_47_750: EngineState: 7F: FF
2021-01-06_14_05_47_750: EngineState: 7F: FF
2021-01-06_14_05_47_750: EngineState: 7F: FF
2021-01-06_14_05_47_755: EngineState: 7F: FF
2021-01-06_14_05_47_755: EngineState: 7F: FF
2021-01-06_14_05_47_755: EngineState: 7F: FF
2021-01-06_14_05_47_755: EngineState: 7F: FF
2021-01-06_14_05_47_756: EngineState: 7F: FF
2021-01-06_14_05_47_756: EngineState: 7F: FF
2021-01-06_14_05_47_759: EngineState: 7F: FF
2021-01-06_14_05_47_759: EngineState: 7F: FF
2021-01-06_14_05_47_759: EngineState: 7F: FF
2021-01-06_14_05_47_759: EngineState: 7F: FF
2021-01-06_14_05_47_762: EngineState: 7F: FF
2021-01-06_14_05_47_762: EngineState: 7F: FF
2021-01-06_14_05_47_763: EngineState: 7F: FF
2021-01-06_14_05_47_763: EngineState: 7F: FF
2021-01-06_14_05_47_766: EngineState: 7F: FF
2021-01-06_14_05_47_766: EngineState: 7F: FF
2021-01-06_14_05_47_767: EngineState: 7F: FF
2021-01-06_14_05_47_767: EngineState: 7F: FF
2021-01-06_14_05_47_767: EngineState: 7F: FF
2021-01-06_14_05_47_771: EngineState: 7F: FF
2021-01-06_14_05_47_771: EngineState: 7F: FF
2021-01-06_14_05_47_771: EngineState: 7F: FF
2021-01-06_14_05_47_772: EngineState: 7F: FF
2021-01-06_14_05_47_777: EngineState: 7F: FF
2021-01-06_14_05_47_778: EngineState: 7F: FF
2021-01-06_14_05_47_778: EngineState: 7F: FF
2021-01-06_14_05_47_778: EngineState: 7F: FF
2021-01-06_14_05_47_779: EngineState: 7F: FF
2021-01-06_14_05_47_781: EngineState: 7F: FF
2021-01-06_14_05_47_781: EngineState: 7F: FF
2021-01-06_14_05_47_781: EngineState: 7F: FF
2021-01-06_14_05_47_781: EngineState: 7F: FF
2021-01-06_14_05_47_783: EngineState: 7F: FF
2021-01-06_14_05_47_784: EngineState: 7F: FF
2021-01-06_14_05_47_784: EngineState: 7F: FF
2021-01-06_14_05_47_784: EngineState: 7F: FF
2021-01-06_14_05_47_786: EngineState: 7F: FF
2021-01-06_14_05_47_786: EngineState: 7F: FF
2021-01-06_14_05_47_786: EngineState: 7F: FF
2021-01-06_14_05_47_787: EngineState: 7F: FF
2021-01-06_14_05_47_787: EngineState: 7F: FF
2021-01-06_14_05_47_789: EngineState: 7F: FF
2021-01-06_14_05_47_790: EngineState: 7F: FF
2021-01-06_14_05_47_790: EngineState: 7F: FF
2021-01-06_14_05_47_791: EngineState: 7F: FF
2021-01-06_14_05_47_794: EngineState: 7F: FF
2021-01-06_14_05_47_794: EngineState: 7F: FF
2021-01-06_14_05_47_795: EngineState: 7F: FF
2021-01-06_14_05_47_798: EngineState: 7F: FF
2021-01-06_14_05_47_798: EngineState: 7F: FF
2021-01-06_14_05_47_799: EngineState: 7F: FF
2021-01-06_14_05_47_799: EngineState: 7F: FF
2021-01-06_14_05_47_800: EngineState: 7F: FF
2021-01-06_14_05_47_804: EngineState: 7F: FF
2021-01-06_14_05_47_804: EngineState: 7F: FF
2021-01-06_14_05_47_804: EngineState: 7F: FF
2021-01-06_14_05_47_805: EngineState: 7F: FF
2021-01-06_14_05_47_809: EngineState: 7F: FF
2021-01-06_14_05_47_809: EngineState: 7F: FF
```

dron0gus, 2021-01-06:
`pin PD5: tle8888 CS`
is get lost after settings flash.

dron0gus, 2021-01-06:
Should be fixed after https://github.com/rusefi/rusefi/pull/2194

rusefillc, 2021-01-06:
@shadowm60 please test?

closed 2021-01-06 by rusefillc

shadowm60, 2021-01-06:
Bench test looks fine, after burn pin function is back online. Thanks