rusefi/rusefi #2112: HW_CO assertWave

rusefillc, 2020-12-22:
At the moment real HW tests are not asserting injection and ignition timing. Open question if we shall mock some sensors to get repeatable numbers

```
        /**
         * with real hardware we have noise on all analog inputs which gives us random sensor data, we cannot really
         * test exact numbers yet
         */
        TestingUtils.isRealHardware = true;
```

mck1117, 2020-12-22:
it's probably worth separating out two types of tests:
- tests that merely check for existence of a particular signal (aka, in wasted spark mode, we should see only half the cylinders being fired)
- tests that mock enough sensors that we check that ign/inj timing is correct

rusefillc, 2020-12-22:
that's a great idea, that would be a simple change for isRealHardware case to check presence

we just had even limited HW CI helping us with defects, and it would be beneficial to check at least some correct timing on real HW. we do need integrated tests since those have proven to be helpful.

maybe runtime-only sensor mocking available via console commands only is a reasonable compromise

mck1117, 2020-12-22:
I think it's also useful to actually write hardware tests that do something that isn't possible or easy in unit tests.  Right now they're all sort of just smoke tests - configure for engine X, then confirm that no fire comes out.