rusefi/rusefi #2123: Wrap PAL driver into universal 'startDigitalCapture'

rusefillc, 2020-12-23:
Need a proper implementation for

```
#if HAL_VSS_USE_PAL
	ioline_t pal_line = PAL_LINE(getHwPort("vss", CONFIG(vehicleSpeedSensorInputPin)), getHwPin("vss", CONFIG(vehicleSpeedSensorInputPin)));
	efiExtiEnablePin("VSS", CONFIG(vehicleSpeedSensorInputPin), PAL_EVENT_MODE_BOTH_EDGES, vsExtiCallback, (void *)pal_line);
#elif HAL_USE_ICU
	digital_input_s* vehicleSpeedInput = startDigitalCapture("VSS", CONFIG(vehicleSpeedSensorInputPin));

	vehicleSpeedInput->widthListeners.registerCallback((VoidInt) vsAnaWidthCallback, NULL);
#else
	#error "HAL_USE_ICU or HAL_VSS_USE_PAL should be enabled to use EFI_VEHICLE_SPEED"
#endif /* HAL_VSS_USE_PAL, HAL_USE_ICU */
```

mck1117, 2020-12-24:
this is already done for trigger - it just needs to get pulled up a level

rusefillc, 2021-07-31:
digital_input_s should be abstract helper class regardless of chosen stm32 PAL=exti OR ICU timer hardware

the idea is the we have two copies of startDigitalCapture() depending on chosen hardware/compilation options, with higher level business logic calling same method and compiler picking chosen hardware implementation based on compilation settings

i.e. business logic does not have conditional compilation but actual startDigitalCapture implementation is compiled conditionally

mck1117, 2021-07-31:
Or startDigitalCapture goes down the list of possible capture options and sees if a pin can do each. Missing a type of capture, doesn't check it.

rusefillc, 2021-11-08:
@racer-coder I've mentioned this one on the phone

rusefillc, 2022-11-01:
Things have changed

closed 2022-11-01 by rusefillc