rusefi/rusefi #2144: create pinout markup based on mapping.md

rusefillc, 2020-12-26:
At the moment we manually maintain two files - markup.md pinout and mapping.yaml for .ini

it should be one source of truth, potentially also generating some C header to be used for engine configs

rusefillc, 2021-01-11:
we have machine-readable https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/proteus/mapping.yaml

which overlaps with manually supported https://github.com/rusefi/rusefi/wiki/Hardware-Proteus-Wiring-v03

i would prefer the table in https://github.com/rusefi/rusefi/wiki/Hardware-Proteus-Wiring-v03 in to be generated based on  https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/proteus/mapping.yaml

 https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/proteus/mapping.yaml would probably need more data to generated section of https://github.com/rusefi/rusefi/wiki/Hardware-Proteus-Wiring-v03

rusefillc, 2021-01-11:
there is also different a bit related https://github.com/rusefi/rusefi_documentation/issues/118 in different GF because reasons.

chuckwagoncomputing, 2021-01-11:
Format proposal:

```
outputs:
  # TLE8888 injector channels
  - id: TLE8888_PIN_1
    pin_number: 37 
    function: Injector 1
    type: inj  
    color: grey right 

  - id: TLE8888_PIN_2
    pin_number: 38 
    function: Injector 2 ```

rusefillc, 2021-01-12:
My name is... (fill in) and I approve this proposal!

rusefillc, 2021-01-12:
drama is to change all mapping files and java code which reads it already in the same PR

mck1117, 2021-01-12:
oh, and we can't really change the existing pin names in the ini, because TS stores them by name, not by binary value

chuckwagoncomputing, 2021-01-12:
The name being e.g. "37 - Injector #1"?
Or TLE8888_PIN_1?

mck1117, 2021-01-12:
the string "37 - Injector #1"

chuckwagoncomputing, 2021-01-12:
I was hoping we could rebuild that in Java, but it looks like not everything follows the same pattern.

rusefillc, 2021-01-12:
because not machine readable thus this ticket :)

chuckwagoncomputing, 2021-01-12:
So I guess we need something like this:
```
outputs:
  # TLE8888 injector channels
  - id: TLE8888_PIN_1
    pin_number: 37
    ts_name: 37 - Injector 1
    function: Injector 1
    type: inj
    color: grey right
```
I don't like the redundant data but I guess it can't be helped without messing up the ini format.

chuckwagoncomputing, 2021-01-12:
And just so we're clear, the redundancy is necessary because of stuff like this:
```
  # J2
  GPIOA_15: "AUX J2 PA15"
  GPIOB_8:  "AUX J2 PB8"
  GPIOB_9:  "AUX J2 PB9"
  GPIOC_12: "AUX J2 PC12"
  GPIOE_0:  "AUX J6 PE0"
  GPIOE_6:  "AUX J10 PE6"
  GPIOE_5:  "AUX J11 PE5"
  GPIOB_7:  "AUX J18 PB7"

  GPIOB_0: "AUX AV8 reuse"
  GPIOB_1: "AUX AV9 reuse"
  GPIOA_4: "AUX AV10 reuse"
  GPIOC_4: "AUX AV6 reuse"
```

rusefillc, 2021-01-12:
I do not follow

.ini is already generated based on mapping.yaml?

chuckwagoncomputing, 2021-01-12:
If I understand what Matt said, yes, and we can't change the names in the ini without breaking user's configs. I hoped we could reconstruct the name string from the old format using the 'pin_number' and 'function' fields, but not all pins follow the same format, especially across different boards. Therefore we need to retain the name string from the old format, in the new format.
The reason we have the pin_number and the function fields separated in the new format is for constructing the table in the md, as it appears now. We could change the format of the table, but I'm already brainstorming something else this yaml could be used for, which would need the pin number as its own field.

mck1117, 2021-01-12:
we could just put a "tunerstudio pin name" in the yaml?

chuckwagoncomputing, 2021-01-12:
> ```
> outputs:
>   # TLE8888 injector channels
>   - id: TLE8888_PIN_1
>     pin_number: 37
>     ts_name: 37 - Injector 1
>     function: Injector 1
>     type: inj
>     color: grey right
> ```

chuckwagoncomputing, 2021-01-12:
I forgot that name and function are separate table columns, so we need even more redundancy:

```
- id: TLE8888_PIN_1
    pin_number: 37
    name: Injector 1
    function: Injector 1
    ts_name: 37 - Injector 1
    type: inj
    color: grey right
```

chuckwagoncomputing, 2021-01-12:
More confusion:
Some ts names, e.g. "23 - AN temp 2" are assigned to multiple pin names, in this case "GPIOA_1" and "EFI_ADC_1", in different sections.

chuckwagoncomputing, 2021-01-12:
This does not map 1:1 with the table, which has only one row for 23.

chuckwagoncomputing, 2021-01-12:
The wiki table is a connector pin map, while the yaml is a MCU pin map. I think we should have a separate yaml for every connector that maps the IDs in mapping.yaml to pin numbers. The MRE for example, would have one for the main connector, and others if we wanted tables for the other connectors, such as J2 and J8.

chuckwagoncomputing, 2021-01-12:
```
outputs:
  # TLE8888 injector channels
  - id: TLE8888_PIN_1
    function: Injector 1
    ts_name: 37 - Injector 1
    type: inj

  - id: TLE8888_PIN_2
    function: Injector 2
    ts_name: 38 - Injector 1
    type: inj
```

```
pins:
  - pin: 37
    id: TLE8888_PIN_1
    color: grey right

  - pin: 38
    id: TLE8888_PIN_2
    color: grey
```

mck1117, 2021-01-13:
Is it worth inverting the structure?  Declare a list of pins, and for each one, state whether it's a [outputs, event_inputs, switch_inputs, analog_inputs] (multiple allowed)?

chuckwagoncomputing, 2021-01-13:
I was so close to thinking of that. :)
```
pins:
  - pin: 1
    id: EFI_ADC_11
    color: white right
    function: 12V Supply from main relay output, see also pin #29
    ts_name: Battery Sense
    type: 12V
    class: analog_inputs

  # default Idle Air Control TLE8888 high current low side: VVT2 TLE8888_OUT5
  - pin: 3
    id: TLE8888_PIN_5
    color: white
    function: Idle solenoid
    ts_name: 3 - Lowside 2
    type: ls
    class: outputs

...

  - pin: 18
    id: [GPIOA_0, EFI_ADC_0]
    color: red
    function: CLT sensor
    ts_name: 18 - AN temp 1
    type: at
    class: [switch_inputs, analog_inputs]
```

I like that the best of anything yet. It'll take some care on my part while converting to not get things switched around, so we can assume that the first element in 'id' goes with the first element in 'class' and so on.

mck1117, 2021-01-13:
or just store the mapping from GPIOx_Y -> EFI_ADC_Z, and only put the GPIO pin in the `id`

mck1117, 2021-01-13:
(that mapping would be constant per-MCU family, all STM32 have the same mapping)

chuckwagoncomputing, 2021-01-13:
How are GPIOA_0, EFI_ADC_0, etc. used in TS? I grepped a generated .ini for them and didn't find them. Is it important to keep them assigned to the correct class?

rusefillc, 2021-01-13:
TS does not care for "GPIOA_0" string, it cares for GPIOA_0 value from enum.

TS index in enum gives you int value

in C .h we have those enums hard-coded

that int is how TS and rusEFI integrate. At the moment the link is so vague you were asking about it!

chuckwagoncomputing, 2021-01-13:
If the mapping is stored in a separate structure, then we have to look up the mapping for every pin in case it exists. Or we could maybe only do it on pins that have multiple values in 'class'. Are you thinking that if it's a separate file, it could be used by multiple boards?

chuckwagoncomputing, 2021-01-15:
Any ideas on how to parse this efficiently? It seems we will have to build everything in memory, then write it to the file after it's all done. Much less efficient then the current setup.

rusefillc, 2021-01-15:
Those files are tiny right?

There must be something simpler than java to handle yaml but I only java :(

chuckwagoncomputing, 2021-01-15:
The yaml for MRE is 8K. It's not as though it will be /slow/, but it will be slower and use more RAM than it has, and I don't like the taste of performance regression.

chuckwagoncomputing, 2021-01-15:
BoardReader contains this comment:
```
/**
 * This tool read mapping yaml file and produces a .txt file with defines in the rusefi.txt format
 *
 * This is a lazy to implement mapping as a separate phase of code generation. Technically this could be merged into
 * the primary generation to avoid the intermediate file.
 */
```

Is $BOARD_prefix.txt included as is anywhere? Or is it only parsed by ConfigDefinition? Getting rid of the intermediate textual representation, and reading the yaml in ConfigDefinition would simplify things.

rusefillc, 2021-01-15:
it's a mess and I barely remember

I believe that only one piece of code reads $BOARD_prefix.txt - so if you see a consumer of $BOARD_prefix.txt please assume that it is the only consumer.

chuckwagoncomputing, 2021-01-20:
This system of mapping by connector pin falls apart a little bit for MCU pins that connect to other chips, e.g. MAX9924 on Proteus. For the documentation table, we need to know the purpose of the connector pin, and for the .ini we need the name/id of the MCU pin. However, these two things don't go together. The connector-purpose matching makes sense in the connector yaml file, but the MCU pin mapping doesn't really make sense there.

chuckwagoncomputing, 2021-01-20:
Option 1:
```
  - pin: 4
    function: Variable Reluctance #2 positive
    type: vr

  - id: GPIOE_8
    class: event_inputs
    ts_name: VR 2

  - pin: 5
    function: Variable Reluctance #1 positive
    type: vr

  - id: GPIOE_7
    class: event_inputs
    ts_name: VR 1
```

Option 2:
Separate file, 'internal.yaml' or something

rusefillc, 2021-01-20:
My head is a bit somewhere else for the next 21 days give or take :(

rusefillc, 2021-02-06:
@chuckwagoncomputing now that we have fancy https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/hellen/hellen72/connectors/main.yaml

can we produce some .md from it?

chuckwagoncomputing, 2021-02-06:
What kind of information are you looking for in the .md?

rusefillc, 2021-02-06:
Ideally I would love to generate the pinout table like at the https://github.com/rusefi/rusefi/wiki/Hardware-microRusEfi-wiring programmatically

I would still have to copy-paste generated .md into some public wiki but at least the table content would be somewhat machine-generated?

rusefillc, 2021-02-06:
@chuckwagoncomputing shall we add "index.html" into all board folders?

at the moment one has to know "main" name and all names of all connectors and I am not sure if we have a place where all connectors are enumerated

chuckwagoncomputing, 2021-02-06:
[image]

chuckwagoncomputing, 2021-02-06:
All this can be yours and more, simply by enabling index on the pinouts directory.

chuckwagoncomputing, 2021-02-13:
Assuming you are satisfied with what I have created, this can be closed.

rusefillc, 2021-02-13:
https://rusefi.com/docs/pinouts/ is _amazing_

thank you one more time

See also https://github.com/rusefi/rusefi_documentation/pull/119

closed 2021-02-13 by rusefillc