rusefi/rusefi #2153: FATAL while bench testing VR

rusefillc, 2020-12-28:
Brand new Proteus potentially with assembly issues
two real VR sensors connector

a chunk of metal to fly over

```
2020-12-27_22_06_22_292: EngineState: CRITICAL error: toothed_previous_time after nowNt 20201227@proteus
2020-12-27_22_06_22_493: EngineState: CRITICAL error: toothed_previous_time after nowNt 20201227@proteus
2020-12-27_22_06_22_694: EngineState: CRITICAL error: toothed_previous_time after nowNt 20201227@proteus
2020-12-27_22_06_22_895: EngineState: CRITICAL error: toothed_previous_time after nowNt 20201227@proteus
2020-12-27_22_06_23_097: EngineState: CRITICAL error: toothed_previous_time after nowNt 20201227@proteus
2020-12-27_22_06_23_298: EngineState: CRITICAL error: toothed_previous_time after nowNt 20201227@proteus
```

rusefillc, 2020-12-28:
looks like this only happens in case of self-stimulation and real VR happening at the same time. maybe not an issue?

mck1117, 2023-01-20:
we now inhibit hardware input in case of self stim

closed 2023-01-20 by mck1117