rusefi/rusefi #2163: Bug in PID/Boost control (Maybe?)

960, 2020-12-30:
Having some issues using closed-loop boost.

Between samples, the readings and output are 0.
Shouldnt it kept the last readings until next sample??

Placed some extra debugchannels in boost control to show.

https://user-images.githubusercontent.com/9406140/103371259-4b7b7580-4acf-11eb-9508-732264bd67b0.mp4

960, 2020-12-30:
This is the first time I have tried to use higher control-rate, so havent seen this before.
With short time like 5ms it's not visible.

mck1117, 2021-01-01:
can you PR the logging you need in to main firmware so we can repro?

mck1117, 2021-01-01:
> Between samples, the readings and output are 0.
> Shouldnt it kept the last readings until next sample??

what does "between samples" mean?

960, 2021-01-04:
This is actually the debug-channels.

It also explains the weird spikes I have seen before with debug-channels.

They seem to update at it's own rate, and go to 0 between.

I added normal channels, and they are nice and steady.

by between samples, I mean the control-period.

mck1117, 2021-01-05:
No, they don't update at their own rate or anything.  You need so show some source code (and PR it!) if you want me to be able to help you.

closed 2021-01-05 by 960