rusefi/rusefi #2197: 2D interpolation for other than float

960, 2021-01-07:
Could we have a template for 2D interpolation of other types than float?'

Like we have for 3D tables

960, 2021-01-11:
Just tested at a curve:

```
./controllers/algo/airmass/airmass.cpp:28:66: error: invalid operands of types 'uint8_t [8]' {aka 'unsigned char [8]'} and 'int' to binary 'operator/'
   28 |   float idleVe = interpolate2d("idleVe", rpm, config->idleVeBins / RPM_1_BYTE_PACKING_MULT, config->idleVe * PACK_MULT_VE);
      |                                               ~~~~~~~~~~~~~~~~~~ ^
      |                                                       |
      |                                                       uint8_t [8] {aka unsigned char [8]}
./controllers/algo/airmass/airmass.cpp:28:108: error: invalid operands of types 'uint16_t [8]' {aka 'short unsigned int [8]'} and 'int' to binary 'operator*'
   28 |   float idleVe = interpolate2d("idleVe", rpm, config->idleVeBins / RPM_1_BYTE_PACKING_MULT, config->idleVe * PACK_MULT_VE);
      |                                                                                             ~~~~~~~~~~~~~~ ^
      |                                                                                                     |
      |                                                                                                     uint16_t [8] {aka short unsigned int [8]}
```

mck1117, 2021-01-11:
the error says exactly what you can't do - you're trying to multiply an array by a scalar

960, 2021-01-11:
But isnt that what we do with the tables bins?

Like all the coolant bins would be nice to offset -40, and scaling the values.

mck1117, 2021-01-11:
`interpolate2d("idleVe", rpm / RPM_1_BYTE_PACKING_MULT, config->idleVeBins, config->idleVe) / PACK_MULT_VE`

closed 2021-01-11 by rusefillc

960, 2021-01-11:
Ah, ok..

Thanks :-)

960, 2021-01-15:
By the way, 3d tables with different sized load axis vs rpm axis doesnt work.

Also, is it possible to use different typedefs at the axis for 3d maps?

mck1117, 2021-01-15:
can you elaborate on "doesn't work"

960, 2021-01-15:
It doesnt return value from right place in map.

I dont remember exactly, but when I swapped LOAD_COUNT / RPM_COUNT in table helper it got better, but still wrong.

Tested at several tables

mck1117, 2021-01-16:
You're going to have to give more detail than that.  What table? Real example please?