rusefi/rusefi #2207: HW CI no longer reliable

rusefillc, 2021-01-09:
it looks like HW CI is not reliable as it used to be?

it looks like the HW CI has started to happen around https://github.com/rusefi/rusefi/pull/2159 but could be a coincident?

rusefillc, 2021-01-09:
http://rusefici.freeddns.org:23456/job/integration_and_primary_bundle/2584/

rusefillc, 2021-01-09:
http://rusefici.freeddns.org:23456/job/integration_and_primary_bundle/2593/
```
    [junit] Testcase: testSachs took 6,674 sec
    [junit] Testcase: testMazdaProtege took 16,952 sec
    [junit]     FAILED
    [junit] Not true: ProtegeLX cranking
    [junit] junit.framework.AssertionFailedError: Not true: ProtegeLX cranking
    [junit]     at com.rusefi.TestingUtils.assertTrue(TestingUtils.java:34)
    [junit]     at com.rusefi.FunctionalTest.assertWaveNotNull(FunctionalTest.java:477)
    [junit]     at com.rusefi.FunctionalTest.testMazdaProtege(FunctionalTest.java:281)
```

rusefillc, 2021-01-10:
http://rusefici.freeddns.org:23456/job/integration_and_primary_bundle/2599/
```
    [junit] 2021-01-09_19_37_49_173<EOT>: AUTOTEST nextChart() in 547
    [junit] ------------- ---------------- ---------------
    [junit] 
    [junit] Testcase: testFord6 took 7,712 sec
    [junit] Testcase: testCustomEngine took 5,422 sec
    [junit] Testcase: testSachs took 5,609 sec
    [junit] Testcase: testRevLimiter took 8,531 sec
    [junit] Testcase: testMazdaProtege took 14,765 sec
    [junit] 	FAILED
    [junit] Not true: ProtegeLX running
    [junit] junit.framework.AssertionFailedError: Not true: ProtegeLX running
    [junit] 	at com.rusefi.TestingUtils.assertTrue(TestingUtils.java:34)
    [junit] 	at com.rusefi.FunctionalTest.assertWaveNotNull(FunctionalTest.java:477)
    [junit] 	at com.rusefi.FunctionalTest.testMazdaProtege(FunctionalTest.java:288)
    [junit] 
    [junit] Testcase: testFordAspire took 45,391 sec
    [junit] Testcase: testCamaro took 5,437 sec
    [junit] Testcase: testCitroenBerlingo took 5,609 sec
    [junit] Testcase: test2003DodgeNeon took 16,516 sec
    [junit] 	FAILED
    [junit] Not true: 2003 Neon running
    [junit] junit.framework.AssertionFailedError: Not true: 2003 Neon running
    [junit] 	at com.rusefi.TestingUtils.assertTrue(TestingUtils.java:34)
    [junit] 	at com.rusefi.FunctionalTest.assertWaveNotNull(FunctionalTest.java:477)
    [junit] 	at com.rusefi.FunctionalTest.test2003DodgeNeon(FunctionalTest.java:254)
    [junit] 
    [junit] Testcase: testChangingIgnitionMode took 8,531 sec
    [junit] Testcase: testMazdaMiata2003 took 5,469 sec
    [junit] Testcase: test1995DodgeNeon took 7 sec
```

rusefillc, 2021-01-10:
revert of https://github.com/rusefi/rusefi/pull/2159 did not help
revert of https://github.com/rusefi/rusefi/pull/2159 was reverted

mck1117, 2021-01-10:
I have a theory:
Scheduling injection requires periodic calls to `FuelSchedule::addFuelEventsForCylinder`.  This happens upon injector close, and once at startup to "prime" the system before the engine starts.
If the trigger pattern changes from one with more teeth to one with fewer teeth, I think there's a case in which the update can get skipped, so some injection channels get stuck with a trigger index that's very high, so those injectors NEVER get opened again, and thus can't reschedule to the correct trigger tooth.

mck1117, 2021-01-11:
update to theory:
ok so here's the bug for neon
- the failure always happens after the switch to "use only rising" trigger mode
- which triggers a rebuild of trigger data
- if there is some very particular timing, injection events can get stuck with a reference trigger tooth that NO LONGER EXISTS
- ...and so they never fire

rusefillc, 2021-01-11:
Got 30+ good runs in a row
Enabled hourly build in addition to git-triggered build

[image]

closed 2021-01-11 by rusefillc

rusefillc, 2021-01-11:
#2751

```
    [junit] I 210111 131655.403 [communication executor1] MessagesCentral - postMessage EngineState: WARNING: Can't connect or mount MMC/SD
    [junit] ------------- ---------------- ---------------
    [junit]
    [junit] Testcase: testVW took 51,039 sec
    [junit] Testcase: testV12 took 91,134 sec
    [junit]     Caused an ERROR
    [junit] rpm change did not happen: 60, actual 123.0
    [junit] java.lang.IllegalStateException: rpm change did not happen: 60, actual 123.0
    [junit]     at com.rusefi.IoUtil.changeRpm(IoUtil.java:96)
    [junit]     at com.rusefi.functional_tests.EcuTestHelper.assertRpmDoesNotJump(EcuTestHelper.java:43)
    [junit]     at com.rusefi.HighRevTest.testV12(HighRevTest.java:29)
```