rusefi/rusefi #2222: Smart Trigger Detector/Generator proposal ("virtual trigger wheel")

andreika-git, 2021-01-11:
[DRAFT]
**Problem Statement**.
The current implementation of the trigger wheel decoder has "trigger errors" concept as a result of non-ideal real trigger signals (interference, wheel/sensor run-out etc.). If one extra tooth is "found" (the false signal is received) or missing, then the whole sync. process can be interrupted (which affects the ignition/injection and engine stability as a whole). But in reality the trigger wheel cannot just disappear or change its shape while rotating, it's the signal that is unreliable. The real wheel cannot instantly stop or accelerate infinitely fast. By knowing that, we can implement a better trigger decoder which would use that knowledge to have more reliable source of crankshaft position for scheduling.

**The proposal**:
* We can spin the "virtual trigger wheel" during the reception of the real trigger signal, and compare both received and generated signals to determine the wheel parameters.
* The trigger wheel has ony thee basic parameters:
  1) The wheel type (tooth/signal shape)
  2) The angular speed (i.e. RPM)
  3) The phase (i.e. synchronization offset).
* These 3 parameters define the generator of the virtual trigger's signal, which can be compared to the real one.
* If the virtual signal doesn't match the real one, then we need to adjust the parameters of the 'virtual wheel' to find the best match
* Therefore we can determine the virtual wheel's parameters matching the real trigger wheel and use them to drive the main scheduler/engine logic of the firmware, treating the virtual generated signal as more stable and reliable source instead of the noisy real signal.
* We can even auto-detect the unknown trigger by spinning multiple 'virtual wheels' at the same time and finding out the best matching one (and filtering out the obvious outsiders).
* We can also reject the wrong signal or diagnose the trigger settings using the "similarity" threshold.

mck1117, 2021-01-11:
Is this different to computing the estimated time that the next trigger tooth should arrive, and considering it in error if it's very early?

rusefillc, 2021-01-11:
this has both "no, let's ignore this" and "c'mon, there must be a tooth here, let's execute scheduled events without physical event"

mck1117, 2021-01-13:
sure - so you compute expected next tooth time, then set a timer for 1.25*nextToothTime.  When the timer fires, you generate a fake tooth but backdate it to the expected time.  If an event comes early, ignore it.  I'm not sure it needs the complex abstraction of a "virtual trigger wheel"

rusefillc, 2021-01-13:
Same here - I love the proposal of ignoring and generating fake but do not get the "spinning" abstraction.

chuckwagoncomputing, 2021-03-18:
#987 related