rusefi/rusefi #2225: some cheap used junkyard Accelerometer

rusefillc, 2021-01-15:
Many modern vehicles have cheap small accelerometers. Shall we integrate via CAN?

TODO part numbers

See also #473

mck1117, 2021-01-15:
iirc many are actually only lateral acceleration + yaw rate, not longitudinal accel

rusefillc, 2021-09-30:
G35 0265005693
Audi 8E0907637A lateral acceleration + yaw rate?

@mck1117 Yaw Rate Sensor YRS 3 https://www.bosch-motorsport.com/media/downloads/yaw_rate_sensor_yrs_3.pdf

This sensor is designed to measure the physical effects of yawing, lateral and *longitudinal* acceleration.
In order to achieve this, the sensor features both a
measuring element for yaw rate and two for acceleration, with one appropriate integrated circuit.

rusefillc, 2021-09-30:
kawasaki imu
INERTIAL MEASURING UNIT

rusefillc, 2021-09-30:
Bosch Bank Angle Sensor 1 275 100 853
0265005876

rusefillc, 2021-09-30:
```
2021-09-30_16_18_36_837: EngineState: CAN_rx 130 304 8 6B 86 0 FF F3 7B 6 EE
2021-09-30_16_18_36_840: EngineState: CAN_rx 140 320 8 0 0 0 0 4 27 16 3
2021-09-30_16_18_36_840: EngineState: CAN_rx 130 304 8 63 7F 0 FF 50 7C 7 24
2021-09-30_16_18_36_841: EngineState: CAN_rx 140 320 8 0 0 0 0 6 29 17 F2
2021-09-30_16_18_36_841: EngineState: CAN_rx 130 304 8 33 77 0 FF 8A 7D 8 9
2021-09-30_16_18_36_841: EngineState: CAN_rx 140 320 8 0 0 0 0 8 23 18 82
2021-09-30_16_18_36_841: EngineState: CAN_rx 130 304 8 E5 75 0 FF 46 7F 9 BF
2021-09-30_16_18_36_841: EngineState: CAN_rx 140 320 8 0 0 0 0 A 21 19 4
2021-09-30_16_18_36_841: EngineState: CAN_rx 130 304 8 B2 76 0 FF D1 81 A E4
2021-09-30_16_18_36_842: EngineState: CAN_rx 140 320 8 0 0 0 0 C 0 1A 1F
2021-09-30_16_18_36_842: EngineState: CAN_rx 130 304 8 EB 7B 0 FF 7C 84 B 4C
2021-09-30_16_18_36_842: EngineState: CAN_rx 140 320 8 0 0 0 0 2 20 1B 7E
2021-09-30_16_18_36_948: EngineState: CAN_rx 130 304 8 D8 7E 0 FF A7 85 C 19
2021-09-30_16_18_36_948: EngineState: CAN_rx 140 320 8 0 0 0 0 4 22 1C B0
2021-09-30_16_18_36_948: EngineState: CAN_rx 130 304 8 BB 7B 0 FF 98 84 D F3
2021-09-30_16_18_36_950: EngineState: CAN_rx 140 320 8 0 0 0 0 6 26 1D 83
2021-09-30_16_18_36_956: EngineState: CAN_rx 130 304 8 D7 77 0 FF 7F 82 E 7E
2021-09-30_16_18_36_958: EngineState: CAN_rx 140 320 8 0 0 0 0 8 25 1E 79
2021-09-30_16_18_36_959: EngineState: CAN_rx 130 304 8 DD 79 0 FF F7 7F F 9D
2021-09-30_16_18_36_959: EngineState: CAN_rx 140 320 8 0 0 0 0 A 20 1F 6
2021-09-30_16_18_36_960: EngineState: CAN_rx 130 304 8 4B 84 0 FF 84 7C 0 78
2021-09-30_16_18_36_963: EngineState: CAN_rx 140 320 8 0 0 0 0 C 20 10 B8
2021-09-30_16_18_36_964: EngineState: CAN_rx 130 304 8 85 89 0 FF C 7A 1 2B
2021-09-30_16_18_36_965: EngineState: CAN_rx 140 320 8 0 0 0 0 E 25 11 C7
2021-09-30_16_18_37_055: EngineState: CAN_rx 130 304 8 94 86 0 FF 4 7A 2 21
2021-09-30_16_18_37_055: EngineState: CAN_rx 140 320 8 0 0 0 0 10 26 12 25
2021-09-30_16_18_37_055: EngineState: CAN_rx 130 304 8 D5 85 0 FF FA 7B 3 B8
2021-09-30_16_18_37_055: EngineState: CAN_rx 140 320 8 0 0 0 0 12 29 13 98
2021-09-30_16_18_37_055: EngineState: CAN_rx 130 304 8 C0 84 0 FF 4C 7D 4 78
2021-09-30_16_18_37_059: EngineState: CAN_rx 140 320 8 0 0 0 0 14 23 14 C
2021-09-30_16_18_37_059: EngineState: CAN_rx 130 304 8 C1 7C 0 FF B7 7D 5 F6
2021-09-30_16_18_37_059: EngineState: CAN_rx 140 320 8 0 0 0 0 16 0 15 B3
2021-09-30_16_18_37_060: EngineState: CAN_rx 130 304 8 33 6D 0 FF F1 7E 6 21
2021-09-30_16_18_37_062: EngineState: CAN_rx 140 320 8 0 0 0 0 2 20 16 FF
2021-09-30_16_18_37_163: EngineState: CAN_rx 130 304 8 10 68 0 FF E5 81 7 A8
2021-09-30_16_18_37_164: EngineState: CAN_rx 140 320 8 0 0 0 0 4 0 17 92
2021-09-30_16_18_37_164: EngineState: CAN_rx 130 304 8 F9 71 0 FF C7 84 8 86
2021-09-30_16_18_37_166: EngineState: CAN_rx 140 320 8 0 0 0 0 2 19 18 8E
2021-09-30_16_18_37_171: EngineState: CAN_rx 130 304 8 2D 7F 0 FF FB 84 9 95
2021-09-30_16_18_37_172: EngineState: CAN_rx 140 320 8 0 0 0 0 4 12 19 18
2021-09-30_16_18_37_173: EngineState: CAN_rx 130 304 8 DC 86 0 FF 2C 83 A D
2021-09-30_16_18_37_173: EngineState: CAN_rx 140 320 8 0 0 0 0 6 13 1A 70
2021-09-30_16_18_37_173: EngineState: CAN_rx 130 304 8 84 84 0 FF F3 80 B C3
2021-09-30_16_18_37_173: EngineState: CAN_rx 140 320 8 0 0 0 0 8 33 1B 11
2021-09-30_16_18_37_272: EngineState: CAN_rx 130 304 8 A8 7D 0 FF 91 7E C 9A
2021-09-30_16_18_37_272: EngineState: CAN_rx 140 320 8 0 0 0 0 A 22 1C B9
2021-09-30_16_18_37_273: EngineState: CAN_rx 130 304 8 AC 7D 0 FF 3D 7C D 8C
2021-09-30_16_18_37_273: EngineState: CAN_rx 140 320 8 0 0 0 0 C 21 1D 75
2021-09-30_16_18_37_274: EngineState: CAN_rx 130 304 8 50 7F 0 FF 5B 7B E 99
2021-09-30_16_18_37_274: EngineState: CAN_rx 140 320 8 0 0 0 0 E 20 1E 1D
2021-09-30_16_18_37_279: EngineState: CAN_rx 130 304 8 6D 7E 0 FF 31 7C F 54
2021-09-30_16_18_37_280: EngineState: CAN_rx 140 320 8 0 0 0 0 10 28 1F 4B
2021-09-30_16_18_37_281: EngineState: CAN_rx 130 304 8 2D 7A 0 FF B1 7D 0 B8
2021-09-30_16_18_37_282: EngineState: CAN_rx 140 320 8 0 0 0 0 12 23 10 7D
2021-09-30_16_18_37_284: EngineState: CAN_rx 130 304 8 A 77 0 FF 34 7E 1 90
2021-09-30_16_18_37_285: EngineState: CAN_rx 140 320 8 0 0 0 0 14 33 11 D1
2021-09-30_16_18_37_382: EngineState: CAN_rx 130 304 8 29 78 0 FF FB 7E 2 DC
2021-09-30_16_18_37_383: EngineState: CAN_rx 140 320 8 0 0 0 0 16 20 12 95
2021-09-30_16_18_37_383: EngineState: CAN_rx 130 304 8 F1 76 0 FF 98 81 3 CB
2021-09-30_16_18_37_383: EngineState: CAN_rx 140 320 8 0 0 0 0 18 22 13 19
2021-09-30_16_18_37_383: EngineState: CAN_rx 130 304 8 E3 74 0 FF DB 84 4 9A
2021-09-30_16_18_37_389: EngineState: CAN_rx 140 320 8 0 0 0 0 1A 29 14 C7
2021-09-30_16_18_37_389: EngineState: CAN_rx 130 304 8 64 7C 0 FF F3 85 5 D2
2021-09-30_16_18_37_390: EngineState: CAN_rx 140 320 8 0 0 0 0 1C 0 15 BC
2021-09-30_16_18_37_391: EngineState: CAN_rx 130 304 8 D3 85 0 FF 23 84 6 78
2021-09-30_16_18_37_392: EngineState: CAN_rx 140 320 8 0 0 0 0 2 20 16 FF
2021-09-30_16_18_37_495: EngineState: CAN_rx 130 304 8 A1 84 0 FF BF 81 7 22
2021-09-30_16_18_37_495: EngineState: CAN_rx 140 320 8 0 0 0 0 4 20 17 E7
2021-09-30_16_18_37_495: EngineState: CAN_rx 130 304 8 F0 7F 0 FF 82 7F 8 F
[... 17 lines cut ...]
2021-09-30_16_18_37_617: EngineState: CAN_rx 130 304 8 B0 75 0 FF FB 83 1 FC
2021-09-30_16_18_37_617: EngineState: CAN_rx 140 320 8 0 0 0 0 4 28 11 F3
2021-09-30_16_18_37_617: EngineState: CAN_rx 130 304 8 69 7E 0 FF 3A 7F 2 E8
[... cut ...]
2021-09-30_16_18_37_835: EngineState: CAN_rx 140 320 8 0 0 0 0 2 20 19 44
2021-09-30_16_18_37_836: EngineState: CAN_rx 130 304 8 DE 6E 0 FF E1 7B A F8
2021-09-30_16_18_37_836: EngineState: CAN_rx 140 320 8 0 0 0 0 4 0 1A 13
2021-09-30_16_18_37_837: EngineState: CAN_rx 130 304 8 60 75 0 FF 89 80 B 77
2021-09-30_16_18_37_838: EngineState: CAN_rx 140 320 8 0 0 0 0 2 12 1B 27
2021-09-30_16_18_37_838: EngineState: CAN_rx 130 304 8 59 70 0 FF 43 8D C AA
2021-09-30_16_18_37_839: EngineState: CAN_rx 140 320 8 0 0 0 0 4 F 1C FE
```

rusefillc, 2021-09-30:
Mitch Minton says F037000002 matches Acceleration Sensor MM5.10

[Bosch_IMU_Generic.zip](https://github.com/rusefi/rusefi/files/7262441/Bosch_IMU_Generic.zip)

rusefillc, 2021-10-01:
https://rusefi.com/forum/viewtopic.php?f=4&t=2140

rusefillc, 2021-10-05:
2016 17 18 19 20 Kawasaki ZX10R OEM Lean Angle Bank Sensor Tip Sensor! Ninja 658

f037000002

```
2021-10-05_16_00_26_942: EngineState: CAN_rx 174 372 8 19 80 0 0 2F 80 5 8A
2021-10-05_16_00_26_942: EngineState: CAN_rx 178 376 8 FA 7F 4 D DA 7E 5 82
2021-10-05_16_00_26_942: EngineState: CAN_rx 17C 380 8 0 0 0 0 C4 9E 5 57
2021-10-05_16_00_26_942: EngineState: CAN_rx 174 372 8 14 80 0 0 2A 80 6 C2
2021-10-05_16_00_26_942: EngineState: CAN_rx 178 376 8 3 80 5 2 D9 7E 6 4
2021-10-05_16_00_26_942: EngineState: CAN_rx 17C 380 8 0 0 0 0 C4 9E 6 70
2021-10-05_16_00_26_942: EngineState: CAN_rx 174 372 8 E 80 0 0 25 80 7 88
2021-10-05_16_00_26_942: EngineState: CAN_rx 178 376 8 0 80 6 8 DC 7E 7 72
2021-10-05_16_00_26_942: EngineState: CAN_rx 17C 380 8 0 0 0 0 C1 9E 7 E4
2021-10-05_16_00_26_942: EngineState: CAN_rx 174 372 8 14 80 0 0 25 80 8 E2
2021-10-05_16_00_26_942: EngineState: CAN_rx 178 376 8 FA 7F 0 0 E2 7E 8 DD
2021-10-05_16_00_26_942: EngineState: CAN_rx 17C 380 8 0 0 0 0 B7 9E 8 12
2021-10-05_16_00_26_942: EngineState: CAN_rx 174 372 8 19 80 0 0 2C 80 9 9A
2021-10-05_16_00_26_942: EngineState: CAN_rx 178 376 8 EF 7F 1 28 E8 7E 9 FE
2021-10-05_16_00_26_942: EngineState: CAN_rx 17C 380 8 0 0 0 0 AF 9E 9 1B
2021-10-05_16_00_26_942: EngineState: CAN_rx 174 372 8 16 80 0 0 34 80 A F5
2021-10-05_16_00_26_942: EngineState: CAN_rx 178 376 8 EA 7F 2 20 E8 7E A CF
2021-10-05_16_00_26_943: EngineState: CAN_rx 17C 380 8 0 0 0 0 A9 9E A 39
2021-10-05_16_00_26_943: EngineState: CAN_rx 174 372 8 11 80 0 0 35 80 B E9
2021-10-05_16_00_26_943: EngineState: CAN_rx 178 376 8 EF 7F 3 20 E1 7E B F
2021-10-05_16_00_26_943: EngineState: CAN_rx 17C 380 8 0 0 0 0 B2 9E B BC
2021-10-05_16_00_27_036: EngineState: CAN_rx 174 372 8 11 80 0 0 38 80 C 3F
2021-10-05_16_00_27_036: EngineState: CAN_rx 178 376 8 FA 7F 4 27 D6 7E C 8C
2021-10-05_16_00_27_036: EngineState: CAN_rx 17C 380 8 0 0 0 0 BD 9E C 69
2021-10-05_16_00_27_036: EngineState: CAN_rx 174 372 8 11 80 0 0 3A 80 D 21
2021-10-05_16_00_27_036: EngineState: CAN_rx 178 376 8 8 80 5 22 D3 7E D BB
2021-10-05_16_00_27_037: EngineState: CAN_rx 17C 380 8 0 0 0 0 B9 9E D 72
2021-10-05_16_00_27_037: EngineState: CAN_rx 174 372 8 11 80 0 0 34 80 E F
2021-10-05_16_00_27_037: EngineState: CAN_rx 178 376 8 11 80 6 26 DA 7E E D9
2021-10-05_16_00_27_037: EngineState: CAN_rx 17C 380 8 0 0 0 0 AC 9E E C4
2021-10-05_16_00_27_037: EngineState: CAN_rx 174 372 8 11 80 0 0 25 80 F 85
2021-10-05_16_00_27_037: EngineState: CAN_rx 178 376 8 E 80 7 21 E2 7E F 9B
2021-10-05_16_00_27_037: EngineState: CAN_rx 17C 380 8 0 0 0 0 A9 9E F 50
2021-10-05_16_00_27_037: EngineState: CAN_rx 174 372 8 11 80 0 0 21 80 0 38
2021-10-05_16_00_27_037: EngineState: CAN_rx 178 376 8 0 80 8 23 E1 7E 0 F6
2021-10-05_16_00_27_037: EngineState: CAN_rx 17C 380 8 0 0 0 0 B1 9E 0 FF
2021-10-05_16_00_27_037: EngineState: CAN_rx 174 372 8 14 80 0 0 24 80 1 98
2021-10-05_16_00_27_037: EngineState: CAN_rx 178 376 8 F2 7F 9 21 DF 7E 1 9D
2021-10-05_16_00_27_037: EngineState: CAN_rx 17C 380 8 0 0 0 0 BA 9E 1 62
2021-10-05_16_00_27_037: EngineState: CAN_rx 174 372 8 14 80 0 0 32 80 2 A2
2021-10-05_16_00_27_037: EngineState: CAN_rx 178 376 8 E4 7F A 29 DD 7E 2 F1
2021-10-05_16_00_27_037: EngineState: CAN_rx 17C 380 8 0 0 0 0 C4 9E 2 4
2021-10-05_16_00_27_037: EngineState: CAN_rx 174 372 8 16 80 0 0 3D 80 3 83
2021-10-05_16_00_27_037: EngineState: CAN_rx 178 376 8 D9 7F B 23 E1 7E 3 2F
2021-10-05_16_00_27_037: EngineState: CAN_rx 17C 380 8 0 0 0 0 CF 9E 3 99
2021-10-05_16_00_27_038: EngineState: CAN_rx 174 372 8 16 80 0 0 32 80 4 56
2021-10-05_16_00_27_038: EngineState: CAN_rx 178 376 8 D9 7F 0 0 F0 7E 4 D2
2021-10-05_16_00_27_038: EngineState: CAN_rx 17C 380 8 0 0 0 0 D6 9E 4 51
2021-10-05_16_00_27_038: EngineState: CAN_rx 174 372 8 19 80 0 0 1B 80 5 A4
2021-10-05_16_00_27_038: EngineState: CAN_rx 178 376 8 E4 7F 0 0 F6 7E 5 FA
2021-10-05_16_00_27_038: EngineState: CAN_rx 17C 380 8 0 0 0 0 CF 9E 5 D7
2021-10-05_16_00_27_038: EngineState: CAN_rx 174 372 8 1F 80 0 0 9 80 6 4B
2021-10-05_16_00_27_038: EngineState: CAN_rx 178 376 8 EF 7F 0 FF F0 7E 6 4B
2021-10-05_16_00_27_038: EngineState: CAN_rx 17C 380 8 0 0 0 0 B7 9E 6 B4

```

rusefillc, 2021-10-05:
Done, we have two accelerations on the gauge and logged!

https://rusefi.com/forum/viewtopic.php?p=42482#p42482

rusefillc, 2021-10-06:
[image]