rusefi/rusefi #2234: new mapping for hellen72

rusefillc, 2021-01-22:
hellen72 is in active testing

We need to convert https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/hellen/hellen72/mapping.yaml into new mapping system

chuckwagoncomputing, 2021-01-22:
Should I include the rusEFI harness colors from https://github.com/rusefi/rusefi/wiki/Mazda-Miata-2003 in the hellen72 yaml?

chuckwagoncomputing, 2021-01-22:
Also, where can I find info on the hellen72 "Extension Port"?

rusefillc, 2021-01-22:
@andreika-git ?
at this point we need something barely good enough but quickly :) since at the moment things are broken hellen72?

andreika-git, 2021-01-22:
I don't understand why some of the mappings were converted to the new system, and some were not. What's the point of breaking existing things in the firmware?

chuckwagoncomputing, 2021-01-22:
I took the naive approach of "ls */mapping.yaml" to see which boards had a mapping.yaml. This missed those boards which had a mapping.yaml in a subdirectory.

chuckwagoncomputing, 2021-01-22:
I need some clarification on pin 4U.
https://github.com/rusefi/rusefi/wiki/Mazda-Miata-2003 says both "NC" and "extra main relay for diode"
The TS name in mapping.yaml is "4U - MAP2/Ign8 (A10)"
Is this a pin that didn't have a connection from the factory but is being used in hellen72?

rusefillc, 2021-01-22:
hellen72 could be used on miata or random v8 engine
looks on miata its NC oem and we use that pin for external map

on random v8 it would he ign8

chuckwagoncomputing, 2021-01-22:
I've run into another issue with that pin. It has two different TS names:
"4U - IGN_8 / MAP2"
and 
"4U - MAP2/Ign8 (A10)"
Can we git rid of one?

chuckwagoncomputing, 2021-01-22:
Actually it looks like that's not the only pin with that problem.
Currently ConfigDefinition only supports one. I could change it to accept an array, but I don't see the benefit.

rusefillc, 2021-01-23:
@andreika-git ^^^?

andreika-git, 2021-01-23:
I don't think we need an array for that, the current version "ts_name: 4U - MAP2/Ign8 (A10)" is totally fine IMHO.

chuckwagoncomputing, 2021-01-23:
This is done.

rusefillc, 2021-01-23:
thank you @chuckwagoncomputing

closed 2021-01-23 by rusefillc