rusefi/rusefi #2237: Need Connector Pictures

chuckwagoncomputing, 2021-01-24:
Could we get some pictures of connector faces?
I would like them placed in firmware/config/boards/<board-name>/connectors/
With the name of the file to match the .yaml file they are a photo of

rusefillc, 2021-01-30:
[image]

rusefillc, 2021-01-30:
[image]

rusefillc, 2021-01-30:
[image]

rusefillc, 2021-01-30:
Life is in the way of MRE

rusefillc, 2021-02-14:
[image]

[image]

MRE photos are not amazing but hopefully acceptable

I need a brighter day or try at new house with Frankenso, so far Frankenso was not acceptable

andreika-git, 2021-02-15:
This is for Hellen81 board pinout:
[image]

rusefillc, 2021-02-19:
[image]

rusefillc, 2021-02-19:
121 would be used on https://github.com/rusefi/hellen121vag which we hope to make open source this Sunday

121 would also be used on https://github.com/rusefi/hellen121nissan

[image]

rusefillc, 2021-02-19:
154 might be useful one day BigProteus or Hellen154_9.7 whatever comes first :)

[image]

chuckwagoncomputing, 2021-03-11:
Closing this. If we need more, we can create an issue for them. Thanks for the contributions.

closed 2021-03-11 by chuckwagoncomputing