rusefi/rusefi #2238: ADC3 should be used in a more universal way than just for the sw knock

andreika-git, 2021-01-25:
Currently only ADC1 and ADC2 can be used for generic analog channels, which limits the number of possible channels on STM32. We chould use ADC3 the same way as ADC1/2.

rusefillc, 2021-01-26:
Looks like we have two very separate use-cases:
"slow" sensors like VBatt/CLT/IAT/Baro can share ADC with knock relatively easily. We shall start by making that possible.

"fast" sensors like MAP/TPS are probably way harder to co-exist with knock, we shall split/postpone that into a separate issue.

mck1117, 2021-01-27:
@stefanst has proposed ADS1015 instead, which is JLC, only $1, and adds 4 ADC channels with no contention for internal MCU resources

rusefillc, 2024-11-08:
https://github.com/rusefi/rusefi/issues/3350 does the job for now, closing

closed 2024-11-08 by rusefillc