rusefi/rusefi #2242: self-stim inverted coil

rusefillc, 2021-01-26:
while this is reported on Hellen I am failing to imagine how this could be a HW issue

here we have self-stimulation which causes first logical channel to be inverted
also note uneven injection while MAP is even
"miata vvt" trigger

[image]

rusefillc, 2021-01-26:
https://rusefi.com/online/view.php?msq=412 is the config

rusefillc, 2021-01-26:
@stefanst also worth noting that "Miata VVT" is a special case - we only self-stimulate crank sensor not VVT cam sensor. While this should not cause this ^^^ result it's definitely a bit of a special case.

stefanst, 2021-01-26:
It's not purely a self-stim problem. The inversion also happens with the jimstim. The signal on the jimstim is fairly authentic for crank and cam (Miata NB1 - non-VVT).
I have not checked for uneven-ness of injection using jimstim.

rusefillc, 2021-01-26:
at this point the priority is on replicating this on non-Hellen hardware in any way shape or form.

rusefillc, 2021-11-07:
I assume this is not an issue

closed 2021-11-07 by rusefillc