rusefi/rusefi #2244: limp mode too good for bench testing

rusefillc, 2021-01-27:
with bench testing high RPM we have limping manager cutting stuff like fuel but there is no clear UI for that

plan to change logic of existing indicators

rusefillc, 2021-01-27:
maybe separate issue maybe same issue:

[test_screwed_up_bin.zip](https://github.com/rusefi/rusefi/files/5877854/test_screwed_up_bin.zip)

@mck1117 attached a Hellen binary which would happily run on Discovery

if self-stim it would show RPM without injection or ignition, I have no idea why. We need to improve usability of error reporting.

rusefillc, 2021-01-27:
Andrey  12:31 AM
float idealGasLaw(float volume, float pressure, float temperature) {
	return volume * pressure / (AIR_R * temperature);
}

New

Matthew Kennedy  12:32 AM
Stefan is right
```
```
```so first we have

float validateMap(float mapKPa DECLARE_ENGINE_PARAMETER_SUFFIX) {
	if (cisnan(mapKPa) || mapKPa < CONFIG(mapErrorDetectionTooLow) || mapKPa > CONFIG(mapErrorDetectionTooHigh)) {
		warning(OBD_Manifold_Absolute_Pressure_Circuit_Malfunction, "unexpected MAP value: %.2f", mapKPa);
		return 0;
	}
	return mapKPa;
}
```
then we multiple by zero

we need to integrate this into limpManager
because at the moment broken MAP kills engine in a very unclear way

stefanst, 2021-01-27:
This has been suggested before: Limp manager should use some crude form of alpha-n for limping. Maybe have a translation table of TPS -> MAP. (2D, or better 3D)?
Or, if the TPS is not available just use a fixed, programmable MAP value. Probably 60kpa or so.

mck1117, 2021-01-27:
> Maybe have a translation table of TPS -> MAP. (2D, or better 3D)?

That table has to be 3d, or if it's 2d, it has to be TPS -> mass flow rate, not pressure.  There are also some transient fuel correction models that need that table anyway, so it might get dual usage.