rusefi/rusefi #2257: Refactor EFI_MAIN_RELAY_CONTROL

rusefillc, 2021-01-31:
Too few people underhand EFI_MAIN_RELAY_CONTROL mode. The "EFI_MAIN_RELAY_CONTROL" name is definitely not helping

* do we need more than one main relay strategy? why?
* if we do, what would be a better name for EFI_MAIN_RELAY_CONTROL?

rusefillc, 2021-01-31:
turns out we actually have three MR inmplementations:

1) EFI_MAIN_RELAY_CONTROL with FSIO
2) EFI_MAIN_RELAY_CONTROL special edition if FSIO is disabled
3) not EFI_MAIN_RELAY_CONTROL

rusefillc, 2021-01-31:
EFI_MAIN_RELAY_CONTROL is primarily about shutdown

andreika-git, 2021-01-31:
We can try:
1) Use EFI_MAIN_RELAY_CONTROL branch as default and remove branches;
2) For LE_METHOD_VBATT, we return sensors.vBatt, not getVBatt().
Then, hopefully, the mainRelayLogic (fsio) will work correctly in all cases:
- if ECU is powered from the ignition switch voltage, then it doesn't matter anyway, the firmware dies if the switch is off;
- if vBatt is tied to the battery voltage, then it means the default relay control behavior (always on);
- if vBatt is tied to the ignition switch voltage, and ECU is powered from the main relay output, then is acts as was intended by EFI_MAIN_RELAY_CONTROL...
- if vBatt is not measured yet (no slow ADC yet), then we have "time_since_boot" protection (which should be renamed to time_since_ignition_switch or smth?)
- if vBatt is not enabled at all, then it falls back to "12" value and the main relay is always on (default);
-

rusefillc, 2021-01-31:
Depends on #2260

rusefillc, 2021-03-11:
we now have #2260 thank you @mck1117 

this issue is on the critical path of Hellen72