rusefi/rusefi #2264: GM/Daewoo 60/2/2/2 Trigger Pattern

hewball, 2021-02-02:
Trigger output from the optical sensors in my GM/Daewoo Distributor on the F8CV Engine

single inside tooth in the manual is referenced as TDC

[image]
[image]

rusefillc, 2021-02-02:
physical position of that single tooth looks a bit different from signal snippet maybe sensors are a bit offset?

open question if I got the relative position of that single tooth right

rusefillc, 2021-02-13:
i assume this is done until someone tells otherwise

closed 2021-02-13 by rusefillc