rusefi/rusefi #2275: mafscaling has a lot of cool stuff?!

rusefillc, 2021-02-03:
mafscaling is an open source of log parsing tools all the way to AFR SD VE tuning

there must be at least something we can learn from there?

https://github.com/vimsh/mafscaling/wiki

chuckwagoncomputing, 2021-02-11:
anything actionable?

rusefillc, 2021-02-11:
@chuckwagoncomputing some research would need to happen

looks like MAF calibration is the easiest of the tools on that set? if not, we shall identify what is

another actionable thing is opening the source code and checking if they separate model from view - i.e. do they hold data in JTable or do they properly hold data separately? that would tells us if command line version of their tool is possible and if we can reuse their code in rusEFI console

overall that's a long shot but we _might_ have a lot to learn from that toolset