rusefi/rusefi #2278: Gather Nissan CAN data

rusefillc, 2021-02-05:
looks like rusEFI build-in CAN sniffer would need to be improved

Abricosvw, 2021-06-23:
All ready something  decoding Nissan can bus data 

https://github.com/jackm/carhack/blob/master/nissan.md

https://github.com/Knio/carhack/blob/master/Cars/Nissan.markdown

rusefillc, 2021-06-25:
Car bus analyzer has a 16 ID limit on software filter :(

We've seen packets - see both list on the left and filter on the right

251,280,284,285,23D,23E,2,2D1,2DE,6F7,794,29E,231,253,2A0,1F9

[image]

rusefillc, 2021-06-27:
We've recoded some CAN stuff

https://github.com/rusefi/rusefi_documentation/tree/master/OEM-Docs/Nissan/2011_Xterra/2011-nissan-CAN-June-2021

rusefillc, 2021-07-07:
https://forum.gurtam.com/viewtopic.php?id=13006
```
Вобщем,удалось считать такие идентификаторы
ID=1F9 (8)   00 00 00 00 00 00 00 00      - 3 и 4 байт*0.125= обороты
ID=215 (6)   FF F0 FF 00 FF FF 00 00
ID=216 (1)   40 00 00 40 C3 C3 00 9E
ID=231 (8)   00 80 00 CA C3 C3 00 9E
ID=233 (4)   00 00 00 40 C3 C3 00 9E
ID=23D (8)   43 00 00 00 00 AD FE 5D
ID=280 (8)   00 00 00 00 00 00 84 40      -5 и 6 байт*const0.01=скорость
ID=2D1 (7)   00 00 00 00 00 EA 02 00
ID=2DE (8)   00 00 80 04 8F FF 07 85
ID=355 (7)   00 00 00 00 00 00 40 40
ID=358 (8)   01 00 00 20 00 00 00 00
ID=35D (8)   C0 03 00 00 00 00 00 00
ID=551 (8)   53 00 00 A0 FF 70 10 00    - младший байт- 45 = температура двигателя
ID=553 (7)   00 00 00 00 00 00 00 9E
ID=5C5 (8)   44 00 A5 4F 00 0C 00 7F     - пробег автомобиля
ID=60D (8)   18 06 00 00 00 00 00 00
ID=625 (6)   32 00 FF 9D 00 00 07 85
ID=6EC (2)   00 00 00 00 00 E8 02 00

Что они означают пока не разобрался
```

mck1117, 2021-07-07:
Translation?

rusefillc, 2021-07-07:
rough translation https://github.com/rusefi/rusefi/commit/f0939aa752a67911da2e29aafe8cdabdedb7005c

rusefillc, 2021-08-06:
https://www.drive2.ru/b/500679938089681452/

rusefillc, 2022-11-10:
https://www.drive2.com/l/530057789272229668/

rusefillc, 2022-11-10:
[Copy of 370Z CAN.xlsx](https://github.com/rusefi/rusefi/files/9980643/Copy.of.370Z.CAN.xlsx)

[Расшифровка CAN .xlsx](https://github.com/rusefi/rusefi/files/9980644/CAN.xlsx)

rusefillc, 2022-11-13:
https://www.drive2.ru/l/558846302222287264/

rusefillc, 2022-11-13:
https://github.com/iDoka/awesome-automotive-can-id#nissan
https://mhhauto.com/Thread-CAN-Database-files-for-Vector-CANalyzer-BMW-Land-Rover-Ford-Nissan-dbc