rusefi/rusefi #2284: H7 compilation

rusefillc, 2021-02-06:
We need H7 in CI

rusefillc, 2021-02-06:
current status:
SPI disabled - need more code in device_mpu_util.h

CAN disabled - differences in driver?

Flash persistence disabled

USB disabled - open question what's the state of USB in H7, ChibiOS code looked suspecious

mck1117, 2021-02-08:
H7 builds and is in CI as of #2317

ADC: disabled - different driver
CAN: disabled - different driver
USB: disabled - different driver?

mck1117, 2021-02-10:
ADC will be helped by #2325

mck1117, 2021-02-25:
Update:
ADC: disabled - different driver
CAN: minor changes needed, but it doesn't work for some reason, see #2393
Flash: disabled - different driver (but appears to only be slightly different
USB: works!

rusefillc, 2021-07-14:
DFU jump https://github.com/rusefi/rusefi/issues/2961

rusefillc, 2021-08-11:
@mck1117 do you have SD card working via SPI on H7? Or for that matter anything on SPI on H7?

mck1117, 2021-08-11:
SD works with a hacked driver on H7.

https://github.com/mck1117/atlas is H7 but uses the SDMMC peripheral instead of SPI since it's lower overhead and doesn't have the silicon bug present in the SPI controller.

andreika-git, 2021-09-20:
@mck1117 What's the current status/plans of "#define EFI_USE_FAST_ADC FALSE" for H7?

rusefillc, 2021-09-20:
@andreika-git is that a rhetorical question? 

looks like https://github.com/rusefi/rusefi/blob/master/firmware/config/stm32h7ems/efifeatures.h#L9