rusefi/rusefi #2305: Configuration flexibility vs Compilation time

dron0gus, 2021-02-07:
Currently we are using USE_SMART_BUILD=yes in Makefile for all boards to speed-up build.
ChibiOS build system parses halconf.h and halconf_community.h with simple
`HALCONF := $(strip $(shell cat $(CONFDIR)/halconf.h $(CONFDIR)/halconf_community.h | egrep -e "\#define"))`
And complies only enabled drivers/modules.
So this build system is not compatible with
1. option pre-difines in files included to halconf.g (stm32f4ems has some HAL_ options defined in efifeatures.h that is included in halconf.h )
2. inheritance of configs from other boards, like it is done for many boards:
`#include "../../stm32f4ems/halconf.h"`

We should either switch back to USE_SMART_BUILD=no, or clean-up all configs.
Clean-up of configs will cause duplication of halconf.h and halconf_community.h for each board.

mck1117, 2021-02-07:
does `USE_SMART_BUILD` actually save time?

mck1117, 2021-02-07:
time with `= no`: 39.5 seconds
time with `= yes`: 36.5 seconds
survey says: no, it doesn't save much time

mck1117, 2021-02-07:
This is also at least tangentially relevant to #2308.  I'd really like to do #2 that you outlined above and make 90% of halconf/mcuconf shared, so that we don't have one copy per MCU family.

PR #2321 (2021-02-11): move chconf/mcuconf/halconf in to hw_ports
progress roughly in the direction of #2282

also fixes #2305 by turning off smart build, so we can now include files from chconf/mcuconf/halconf and it will actually work
closed 2021-02-11 by rusefillc