rusefi/rusefi #2310: unit_tests linking time

rusefillc, 2021-02-07:
unit tests linking time on my pretty new 12 core ssd Dell is unacceptably high. this is a major impediment for TDD

maybe there are tricks we can apply to improve?

gnu hash not SUVs hash? something about string table compression? anything else?

rusefillc, 2021-02-07:
lld says to work with gcc 9+?

https://stackoverflow.com/questions/29361801/is-the-lld-linker-a-drop-in-replacement-for-ld-and-gold

mck1117, 2021-02-07:
I think the biggest gain will come from using not-Windows.  NTFS performance is terrible for compilers.

Some data:
I ran `time make -j18` in the `unit_tests` folder after doing a clean.

GCC, Windows Subsystem for Linux (aka linux on windows, sort of), Core i7-8700k: 1 minute 24 seconds
2019 Macbook Pro, Core i7-9750H: 1 minute 21 seconds (mind you this one is clang, not gcc)
Ubuntu 20.20, Core i7-4790k: 1 minute 8 seconds

So by all accounts, the slowest machine here actually built the fastest.  That's how much ass ext4 kicks as a filesystem.

Ok ok, I get it, I did a full compile, not just a link.  So now let's edit one trivial file (`test_limp.cpp`, because I had it open) and repeat.

WSL, 7800k: 12 seconds
MBP: 3.9 seconds
Linux, 4790k: 6.7 seconds

So we see two things:
- not windows is fast
- clang is fast

rusefillc, 2021-02-08:
actually my issues is not linking at all? make takes 3 MINUTES to do NOTHING?!

```
21:54:30 **** Incremental Build of configuration Default for project unit_tests ****
make -j12 all 
No CCACHE_DIR
make: Nothing to be done for 'all'.
21:57:22 Build Finished. 0 errors, 0 warnings. (took 2m:51s.953ms)
```

while firmware is way faster
```
21:54:18 **** Incremental Build of configuration Default for project firmware ****
make -j12 all 
No CCACHE_DIR
PROJECT_BOARD: st_stm32f4
PROJECT_CPU:   ARCH_STM32F4
CONFDIR:       ./config/stm32f4ems
LDSCRIPT:      config/stm32f4ems/STM32F407xG.ld
make: Nothing to be done for 'all'.
21:54:29 Build Finished. 0 errors, 0 warnings. (took 10s.202ms)
```

rusefillc, 2021-02-08:
well, it's BOTH "nothing" time and linking time issues
@chuckwagoncomputing @dron0gus @mck1117 

[image]

rusefillc, 2021-02-08:
moving into native Linux home drive has IMPROVED PERFORMANCE BIG TIME

[image]

closed 2021-02-08 by rusefillc