rusefi/rusefi #2318: unit_tests under eclipse complain about

rusefillc, 2021-02-08:
both cygwin and WLS setups complain about same libgcov profiling error 

what is this about? could it be not present same as on command line?

[image]

mck1117, 2021-02-08:
this isn't about under eclipse - this happens on linux

rusefillc, 2021-02-08:
actually outside of Eclipse under cygwin rusefi_tests.exe produces this

[image]

so far only way this does NOT happen is WLS 2 command line

[image]

mck1117, 2021-02-08:
run tests, then change something, compile, then run again - you should see it

mck1117, 2021-02-08:
also, those are DIFFERENT errors you're seeing in eclipse vs. cygwin shell

rusefillc, 2021-02-09:
"Version mismatch - expected 10.2 (release) (B02*) got 9.2 (release) (A92*)" might be worth reading actually, looks like i shall bump gcc to 10.2 to match something