rusefi/rusefi #2328: Hardware: board identification

rusefillc, 2021-02-10:
more boards we have higher is the risk of user flashing wrong firmware

it would be great for firmware to be able to validate what well know board it is running

maybe PCF8575 with pull-ups/pull-downs hard-coded differently on each board?

https://jlcpcb.com/parts/componentSearch?searchTxt=PCF8575

OrchardPerformance, 2021-03-06:
With Matt's addition of the .ini into a FAT partition can we use that as an ID for console? 
Check the .ini onboard matches the firmware before the update is applied?
Would not work for the first time flash though.

rusefillc, 2021-03-06:
console means human

My interest is for firmware to validate itself against hardware without the human

OrchardPerformance, 2021-03-06:
You mean in tuner studio?
I was thinking a press of the consoles "update firmware" would initiate the check before update