rusefi/rusefi #2331: Fatal error when setting Analog Volt 5 as pedal 2 on proteus

rmracingelectronics, 2021-02-11:
When setting the 2nd pedal sensor as analog volt 5 on a proteus (only hardware i had to test) there is a fatal error that bricks USB connection and stops tunerstudio form burning the configurations, if repeated a few times it bricks the ecu to a point where it needs firmware upload to unbrick.

rusefillc, 2021-02-12:
@mck1117 did you say you were able to reproduce this? what are steps for me to reproduce locally?

mck1117, 2021-02-12:
I got it to repro once, by loading [this tune](https://rusefi.com/online/view.php?msq=423) then setting the secondary pedal sensor to analog volt 5.

mck1117, 2021-03-14:
Looks like it happened again: a bin is attached to this post.
[brickedfw.zip](https://github.com/rusefi/rusefi/files/6136331/brickedfw.zip)

rmracingelectronics, 2021-03-14:
With me it happens all the time andni tried all sorts of different configurations and settings, I can't explain what triggers that because obviously I don't know the firmware well enough

mck1117, 2021-03-14:
Figured it out.  It's a buffer overflow out of the buffer where we write the error message.

PR #2456 (2021-03-14): fix error/warning buffer overflow
We had no protection against overflowing the error or warning buffers.  If you hit certain errors, it would overflow and stomp whatever memory lay after the error buffer.

This does exactly what was suggested by the todos in code: just use `chvsnprintf` instead of a custom memorystream implementation.

fixes #2331
closed 2021-03-14 by rusefillc