rusefi/rusefi #2368: Launch control doesn't work by input or always on

rmracingelectronics, 2021-02-17:
can't make the launch control to work either by activated by clutch input or aways ON
I tested the default options that look good enough and nothing happens, the engine always revs above the set launch rpm

shadowm60, 2021-02-17:
Can you share config and used fw version? Is the tune online?

rmracingelectronics, 2021-02-17:
[image]
here it is the configurations and firmware version, not that it really matters because I also tried other version and also doesn't work, neither changing any configurations

shadowm60, 2021-02-17:
One quick note is speed is 0, so it will be deactivated, 
Here are my settings that i use.

https://rusefi.com/online/view.php?msq=417

rmracingelectronics, 2021-02-17:
[image]
thanks a lot for that, so it is another case of not using the best wording on TS
''always active disable by speed'' even when not using speed, using clutch input or digital input you have to set a speed if you leave it at zero doesn't work.
so a setting that has nothing to do with the option sellect (if im not using speed why i have to set a speed?) will make it not to work.
the logic on the input is also inverted, if you use a digital input because it has a pull up it becomes always active and only disables with GND

shadowm60, 2021-02-17:
So if using Always Active Disable by speed -> we need a speed threshold to stop limiting engine RPM
https://i.imgur.com/9dTiZXh.png

In case of clutch or button, Speed trashed is not used.

I never used other ECUs then rusEFI so i do not know how to organize them to be more user friendly, probably in case of switch or clutch the input box for speed should be disabled, the whole config window is created way back, and just part of it used.

There is a howto, i do not know how useful it is, but inputs are welcome, and we can make it better, also for testing, if you have inputs, we can apply them.
https://github.com/rusefi/rusefi/wiki/HOWTO-Launch-Control

shadowm60, 2021-02-21:
@rmracingelectronics any updates?

rusefillc, 2021-11-16:
@rmracingelectronics any updates?

rusefillc, 2022-03-28:
closing since no feedback :(

closed 2022-03-28 by rusefillc