rusefi/rusefi #2385: map 121 pin

rusefillc, 2021-02-23:
(not looking for volunteers right now I hope to ask someone for help)

need to map all 121 pins

[image]

https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/hellen/hellen121vag/connectors/main.jpg

https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/hellen/hellen121vag/connectors/main.yaml

rusefillc, 2021-02-23:
[image]

mck1117, 2021-06-29:
this will be of some help: https://github.com/rusefi/hellen121vag/blob/main/boards/hellen121vag-a/board/hellen121vag-a-schematic.pdf

rusefillc, 2021-06-29:
maybe CAM is the one missing?

rusefillc, 2021-06-29:
CAM has stm32 not board names

https://files.slack.com/files-pri/T4NJHQ8QZ-F026UGDUM0R/received_872360346686658.jpeg

mck1117, 2021-06-29:
Crank too

BeerMoneyMotorsports, 2021-06-29:
Pins 82(Crank - ) and 90(Crank + ) come into this module below then it outputs crank in to GPIOB_1.  Im not sure how to map these pins correctly?

[image]

[image]
[image]

BeerMoneyMotorsports, 2021-06-29:
I do not see an issue with the Cam pin mapping.  

[image]

rusefillc, 2021-06-29:
stm32 has one digital pin for crank signal
121 board connector has two pins for VR

to have both 121 board pins on the pinout le't keep both crank+ and crank- in the mapping

to have a nice stm32 name for the digital pin (PB1 you are saying?) let's have this one on crank+

and crank- would live without stm32 pin

BeerMoneyMotorsports, 2021-06-29:
@rusefillc This is how i set it in the yaml. 

[image]

rusefillc, 2021-06-29:
function: Crank Sensor
function: IN_CRANK-

that's inconsistent

overall I am lost

BeerMoneyMotorsports, 2021-06-29:
@rusefillc  Im not sure why it is not showing up correctly in TS?  Even with that inconsistency, it should show up "82 - VR".  Im proposing these changes below to test.  

[image]

rusefillc, 2021-06-29:
We are now blocked by https://github.com/rusefi/rusefi/issues/2880 :(

rusefillc, 2021-06-29:
also I doubt that analog_inputs would help here :(

BeerMoneyMotorsports, 2021-06-29:
The analog addition is to be in sync with how it is done in other yamls for the crank.   It is something i missed when first building this yaml.

mck1117, 2021-06-29:
I don't think we want or need the analog option there.

rusefillc, 2021-10-02:
looks like mostly done

closed 2021-10-02 by rusefillc