rusefi/rusefi #2387: is gen_config invoked too often now?

rusefillc, 2021-02-23:
is looks like we now generate new signature on _any_ git push? we used to have some caching mechanism right?

[image]

chuckwagoncomputing, 2021-02-24:
I thought we decided to do this so that when debugging someone that was using a git build rather than a release, we would be able to tell which version exactly they had.
I'm not here to argue that that is a good reason to do things this way, I'm just saying that if my memory isn't playing me a trick, this was done intentionally.

rusefillc, 2021-02-24:
I do not trust my memory, I do not have any recollection of that.

I am more or less OK either way.

rusefillc, 2021-02-24:
@andreika-git @mck1117 what's your take on this?

mck1117, 2021-02-25:
whether it's intentional or not, the current method is pretty bad - we're committing a new `cache.zip` after every commit, which increases the repo size by 1.1 megabytes, after EVERY SINGLE COMMIT.  Git doesn't diff binary files, it just stores the whole damn thing.

rusefillc, 2021-02-26:
@chuckwagoncomputing do you know which exact change has switched the behavior?

On the one hand we do not use the cache but also we update the cache and repo growth is a real issue. We have to move one way or another but move :)

mck1117, 2021-02-26:
We could also just diff the actual output files against the checked in files, and check if anything other than the timestamp header changed.  That's all we're using the cache for, right?

rusefillc, 2021-02-26:
if we always push _at the moment_ there is no reason to push _cache_ and use _cache_

but I am leaning towards returning it back to when signature was only touched when important files were touched. the value of artifacts having unique signature seems lower at least than the cost of constant recompile for me.

chuckwagoncomputing, 2021-02-27:
Then we need to define what files are important/some heuristic to determine whether a push to master is important.

chuckwagoncomputing, 2021-02-27:
As far as I can tell, this has been the behavior since the initial commit of gen-config.yaml

rusefillc, 2021-03-06:
@chuckwagoncomputing a while back we only had generation sometimes

[image]

rusefillc, 2021-03-06:
[image]

rusefillc, 2021-03-06:
I believe that the selected section is where things got broken @chuckwagoncomputing ?

before this section, we did not have any commit re-generated config

after this section any change has started to generate everything?

interestingly no tools were changed around here? could it be that some unusual change to rusefi_...txt has broken caching mechanism?

[image]

rusefillc, 2021-03-06:
kind of relevant issue - code generator has to use same timezone regardless of who is invoking it where

```
LazyFile controllers/generated/signature_all.h: Not same at 63
file       _DATE 2021.03.06#def
newContent _DATE 2021.03.05#def
```

note that git has March 6 while I still have March 5th

rusefillc, 2021-03-06:
Something is not happy at least for local gen_config.sh invocation @andreika-git 

```
Check the input/output TS files:
Unzip tunerstudio/generated/cache.zip: ./integration/rusefi_config.txt failed!
* the file integration/rusefi_config.txt is changed!
Check the input/output other files:
Unzip tunerstudio/generated/cache.zip: ./integration/rusefi_config.txt failed!
* the file integration/rusefi_config.txt is changed!
CRC32 from integration/rusefi_config.txt = 1787632648
```

rusefillc, 2021-03-06:
@andreika-git I was getting "-1" in 
```if (zippedName.equals(searchName) && zipEntry.getSize() >= 0) {```

They say that the API is way more weird than one would expect, zipEntry.getSize() works totally not the way one would expect.

See https://stackoverflow.com/questions/25602406/zipentry-unknown-size-although-set for details

Now pushing a bit different code to read zip entries

mck1117, 2021-03-13:
looks like this change broke @dron0gus's F7 EG33 board - commits to the corresponding `rusefi_generated.h` stopped then

mck1117, 2021-03-13:
last successful auto commit was 2ecfabd28e7e0ca1a44121be77bd74fb4ed4321d so it was something after that

mck1117, 2021-03-13:
reverting 06e05d821b62e2d6912d6c946ac1483229eaf53e locally gives a different result for EG33 - including the missing defines in `rusefi_generated.h`

rusefillc, 2021-03-13:
reverting https://github.com/rusefi/rusefi/commit/06e05d821b62e2d6912d6c946ac1483229eaf53e is a dead end - without https://github.com/rusefi/rusefi/commit/06e05d821b62e2d6912d6c946ac1483229eaf53e it's just regenerating always without any caching

with https://github.com/rusefi/rusefi/commit/06e05d821b62e2d6912d6c946ac1483229eaf53e caching works for all configurations but does not work right for eg33. I believe cache was not initialized properly for eg33 and that state is not handled properly.

extra logging added, let's see what kind of output we get now

closed 2021-03-13 by rusefillc