rusefi/rusefi #2390: MRE 0.3 F7 board fata error with CPU temp error

shadowm60, 2021-02-24:
Had this error today while driving at multiple times, i could not connect to anything, so it is not something that could be easily reproduced, most of the time happened while idle-ing or near idle rpm's.

[image]
[image]
 
tune: https://rusefi.com/online/view.php?msq=432
Log: <tbd> issues uploading due to bad connection

Reported error code in log: OBD_PCM_Processor_Fault = 606,

rusefillc, 2021-02-24:
@shadowm60 please try latest firmware see change by @mck1117

shadowm60, 2021-02-24:
Ok, i'll flash it in a few hours

shadowm60, 2021-02-25:
@rusefillc it is working perfectly, 40 minute drive, no issues, in the same route i had 5 fails with the old sw. So from my point of view we can close this.

mck1117, 2021-02-25:
excellent - glad it fixed it!

rusefillc, 2021-02-25:
👍 thank you for reporting back

closed 2021-02-25 by rusefillc