rusefi/rusefi #2395: SD card doesn't work on H7

mck1117, 2021-02-24:
USB mass storage enumerates, but you can't actually read/write the volume

smells like something is wrong with DMA or cache

mck1117, 2021-02-24:
spiReceive is reading an extra byte at the beginning - switching from spiReceive to spiReceiveSmall shifts the whole response over by one byte, and the SD card works properly

mck1117, 2021-02-24:
[image]

rusefillc, 2021-02-24:
and whyyyyyy?

mck1117, 2021-02-24:
currently reading the silicon errata for the h7...

mck1117, 2021-02-24:
Woohoo, silicon defect!

[image]

Adding a hardware reset to the MMC SPI driver just before the DMA spiSend/spiReceive calls seems to fix it, and I can read/write files without issue.

rusefillc, 2021-05-13:
@mck1117 close or not?

andreika-git, 2021-08-12:
Do we have a plan to fix it in master?

andreika-git, 2021-08-12:
@mck1117 Maybe we should prepare our Chibios patch for SPI reset to fix MMC?

rusefillc, 2021-09-20:
@andreika-git there is progress here ^^^

PR #3249 (2021-09-20): update chibi for h7 fix
fix #2395
closed 2021-09-20 by rusefillc