rusefi/rusefi #2403: antilag and anti-lag

rusefillc, 2021-02-26:
users are asking for antilag logic in rusEFI

mmatousek89, 2021-07-22:
overview:
when driver demand via PPS is below some threshold
open ETC to 10-40%
retard ignition to 5-30deg ATDC
add 2-20% fuel

hints:
antilag should run engine in neutral gear around 3500rpm
antilag should not add too much torque to crankshaft between 3500-15000rpm driver expect neutral engine behaviour, engine braking is highly desired but hard to obtain

closed loop ideas:
watch turbospeed - open/close ETC to reduce or add turbospeed

conditions:
intake air temperature below 55degC
EGT below 975degC
coolant below 105degC
ALSactive no more than 5000ms in one continous activation
pedal position sensor over 90% at least once in actual driving cycle
als arming switch activated
pedal position sensor below 1%

pnv57, 2021-08-30:
Another thing to note is that rolling antilag would be useful in drag-racing scenarios. The necessary addition would be to maintain via closed-loop control a set RPM (with gear sensing?) or vehicle speed. A possible approach would be to use a modified rev-limiter function I think.

elmepro, 2022-07-20:
Im ready to test antilag if needed, for this season i have (test turbo engine) for drifting

TurboMarian, 2022-09-06:
I really need ALS function for my driftcar. I have car to test it also I would share logs and videos. Maybe bounty will help to implement that function?

Fera31, 2022-09-19:
hi iam from argentina
i work in a rally team i am a tuner
Any information you need about the anti lag system. I can provide it.
we work with MoTeC m800, evo IX and X
MoTeC m130. and m142 direct injection THP engine
I would like to see the addition of the anti lag system in rusefi
I am currently using speeduino with its own antilag system.
I don't know how to modify the rusefi code. but I would like to learn to be able to help

rusefillc, 2022-09-19:
Thank you for your interest in rusEFI!

At the moment we do not have resources to figure out the roadmap towards anti lag feature. https://patreon.com/rusefi is one of the ways to help rusEFI!

TurboMarian, 2022-09-19:
> hi iam from argentina i work in a rally team i am a tuner Any information you need about the anti lag system. I can provide it. we work with MoTeC m800, evo IX and X MoTeC m130. and m142 direct injection THP engine I would like to see the addition of the anti lag system in rusefi i am currently using speeduino with my own anti lag system I don't know how to modify the rusefi code. but I would like to learn to be able to help

I'm working on ALS for rusEFI. I just started and I only work in my spare time so I don't know how long it will be before it's ready.  
[image]

TurboMarian, 2022-09-19:
I also wrote a simple LUA script for ALS. In the video, it was tested on Idle Valve, so not enough air, but as you can see it works.

https://user-images.githubusercontent.com/73337486/191101892-fec019b6-92da-4664-994d-f5b3292be98f.mp4

Fera31, 2022-09-20:
> I'm working on ALS for rusEFI. I just started and I only work in my spare time so I don't know how long it will be before it's ready. [image]

Excellent. I would like to be able to use rusefi and its antilag mode. it is the only thing that stops me from using rusefi ecu

the boost control in als mode. controlled by adding fuel
fuel vs map
that is, the extra fuel reduces the boost and reduces the temperature of the combustion chamber
just the opposite happens with direct injection (be careful)

As for the ignition, it is perfect, maximum -40
it is necessary to add a partial ignition cutoff rpm vs tps

to avoid problems when tuning the antilag. since it is very aggressive for the engine

in the evo IX the accelerator is open fixed between 20% and 30%
The system has an input button for als activation.
and the engine is in idle mode with a constant limiter at 1500 rpm. with partial ignition cut

https://user-images.githubusercontent.com/66860635/191274993-32b34b2f-9376-4c48-b38b-20eb103fc1a9.mp4

if necessary I can share some screenshots of my tuning

rusefillc, 2022-09-21:
@TurboMarian let me know if I can help with any code questions

Do you want to start PRing small pieces maybe the configuration part of the change?

rusefillc, 2022-12-16:
@TurboMarian is your work available anywhere on github?

@Fera31 @idi-moe @elmepro see below does it look like minimal set of settings?

[image]

Who has what test mule? Please see the issue from my standpoint: I have very little feedback from end users, I do not know who has what successful install available for what testing. A lot of amazing progress could happen if we find a way to start helping each other :)

TurboMarian, 2022-12-16:
@rusefillc ALS code is almost done. All the functions are working apart of "ETB AIR ADD" but I will work on it this weekend. 

Should I post it on my github and send you a link or make pull request?
It will be nice if that can be part of rusEFI code.

[image]

rusefillc, 2022-12-16:
Yes would be amazing if you could push to your github and maybe allow me to start picking pieces?

TurboMarian, 2022-12-16:
Sure. I will update ETB air add code soon. 

https://github.com/TurboMarian/rusefi

rusefillc, 2022-12-17:
@TurboMarian what is your forum username? can you please DM me your whatsapp or something? I think we should have a screen sharing session on how to approach the merge

rusefillc, 2022-12-19:
@TurboMarian please blink if Santa is distracting you 🎅

rusefillc, 2022-12-19:
@TurboMarian just emailed about a rebase to make my job easier. My preferred GIT client is what build into java IDE "jetbrains IDEA" it makes complex git operations really easy, please let me know if we can screen share for me to show you.

rusefillc, 2022-12-22:
@TurboMarian please review my picks into master, see also https://rusefi.com/forum/viewtopic.php?p=46858#p46858

@Fera31 @idi-moe @elmepro do you guys use idle solenoid or ETB?

rusefillc, 2022-12-22:
[image]

closed 2022-12-27 by rusefillc

Fera31, 2023-01-06:
@rusefillc  @TurboMarian 
Sorry for the delay
and sorry for my bad english
I am currently using my own speeduino code
with DBW with a separate module. I have test cars but I don't have ecu rusefi,. I would love to start using it. I just don't know how to compile the code or make adjustments. (I would greatly appreciate someone's help to achieve it)
In speeduino I have my own variation of the code with many racing options.

Below I will detail the use of complete antilag in rally cars so as not to have problems and that it works correctly as far as I am concerned according to the MOTEC ecus.

the antilag system
** activation button
Must have power off control
** Maximum Engine coolant temp (when this temperature is exceeded the als must turn off)
** shut down time (normal 5 to 6 sec) (when this time is exceeded the als should shut down)
** TPS MAX ARM (TPS position that must be exceeded for the arming of the antilag) (this avoids the waiting time between activation and activation. The control of the waiting time for arming the als is on the driver's foot)
** TPS MIN ARM (below this limit the als is active (this should vary if it is Electronic throtle body (servo motor) or Mechanic throtle body) if it is mechanical the reading is from the tps. If it is electronic the option can be deactivated or simply read pedal
**shut down exhaust temperature (if ecu has exhaust temperature reading)
** anti lag engine speed limit (maximum rpm with anti lag active)

**Table of (throttle aim main) (throttle servo opening)
this can also be a fixed value

** air bypass valve control if the throttle is mechanical

**Table of (TPS VS RPM) ignition delay of -60° maximum (-40° is a safe limit) (-20 -25 is a normal) (this depends on the torque delivered by the engine)

(this is used in indirect injection)
**Table of (MAP VS RPM) enrichment to control turbo pressure. Adding extra fuel reduces turbo pressure and lowers combustion chamber temperature.
(this can also be controlled using an anti lag boost aim table that directly impacts boost control with anti lag active)

(this is used in indirect injection)
**Table of (TPS VS RPM) partial fuel cut alternating the loss of fuel by cylinders by cycles

**Table of (TPS VS RPM) ignition partial cut alternating the loss of spark by cylinders by cycles

***IMPORTANT . THE CONTROL OF THE TABLES SHOULD ALTERNATE FROM THROTTLE BODY SENSOR TO THROTTLE PEDAL DEPENDING ON THE SYSTEM USED
[image]
[image]
[image]
[image]

TurboMarian, 2023-01-06:
@Fera31 Hello! We have almost all the ALS features from your post alrady implemented in rusEFI code. 

How to compile rusEFI firmware: https://rusefi.com/forum/viewtopic.php?f=5&t=9 

If you have any questions please send me private message. I'd be more than willing to assist.

Fera31, 2023-01-07:
@rusefillc  i am from argentina , i cant get the hardware now
I can make a user and send you messages to the forum,
I do not have a Facebook account

thank you very much for your time and help

@TurboMarian 
thank you. Next week I will try to understand and compile the code. I should get a controller too which I'm trying to do

thank you very much for your time and help

mi-hol, 2023-04-02:
> compile the code. I should get a controller too

@Fera31 did it work?

commit 17a8427c (2022-12-27): Antilag fix #2403