rusefi/rusefi #2404: Boost control by gear

rusefillc, 2021-02-26:
Users are asking for boost control by gear

Benjamin-neo, 2021-08-31:
Will be a very good idea, boost control by gear, with for example button activation. It's like second boost map in better.
In the multi map world , it's the most important , have less boost in some condition ( rain, daily driving, if we let our wife driving....)

rusefillc, 2023-04-25:
@ElDominio how does gear comes into boost?

a 1x6 table of anything? a 32x6 table of anything?

ElDominio, 2023-04-25:
Best way IMO is to have (number of gears)x1, 2D table that applies a correction factor per gear to duty cycle if open loop, or an adder that changes boost target if closed loop

Dizzle159, 2025-07-08:
If said car doesn't have have a gear indicator how would boost by gear even work. I personally think I should be based off vss if there isn't any way of the car knowing what gear it is in.