rusefi/rusefi #2409: torque estimation for external TCU

rusefillc, 2021-02-27:
external TCU love receiving torque value from ECU

shall we start by taking a value from a table by load and rpm?

OrchardPerformance, 2021-02-28:
OEMs do it via a LOT of calibration, measuring of parasitic torque of components, long term durability and friction testing, lube characterization and soooo much work on the software calculating the torque

How do we do it? I think that is not something we can realistically do.
However....
I think we can skirt the issue with no loss of results with regard to controlling a transmission or hybrid motor by simply having a table of torque based on RPM and whatever load is desired
This table can simply be populated with estimated figures and tuned manually or populated while the car is on the dyno.

OrchardPerformance, 2021-02-28:
The other thing we may be able to do is estimate torque based on the bore/stroke and the Kg/h airflow, MAP pressure and a few other abstractions. 
Trouble with doing that is that we cannot take account of the effect of spark angle, compression ratio, combustion efficiency, parasitic losses etc, etc. 
That means it would only ever be an idealized value and likely some way off target at times without careful calculation or validation against measured torque for each install.

rusefillc, 2024-02-17:
starting with ideal engine torque

https://hpwizard.com/engine-horsepower-calculator.html

rusefillc, 2025-08-10:
@ElDominio see https://github.com/rusefi/rusefi/pull/8091