rusefi/rusefi #2410: write saleae analog file

rusefillc, 2021-02-27:
We have the code to write saleae digital files

We shall write saleae analog files as well, making rusEFI a tiny oscilloscope?

rusefillc, 2021-02-27:
a step would be to decipher saleae analog file format

a step towards deciphering would be looking at a few examples :) 

can anyone with a unit attach a few sample files? a very short one with nothing? a longer one with something? etc?

rusefillc, 2021-02-28:
Shaun Veilleux  2:35 PM
Maybe, I set a reminder for Monday. I will see if any are available for use.

Shaun Veilleux  2:45 PM
Question, I dont fully understand what your goal here is. You want to sample some waveform using rusEFI. Save that data in some .bin file to view in Saleae's software?
2:46
If so Saleae shows how to do this here https://support.saleae.com/faq/technical-faq/data-export-format-analog-binary I am sure you have seen this already though so that is my confusion.

Shaun Veilleux  4:10 PM
err, sorry this one is more like it. https://support.saleae.com/faq/technical-faq/binary-export-format-logic-2

Andrey  9:20 PM
@Shaun Veilleux please record all valuable comments onb permanent GH ticket
9:21
with digital .logicdata file format was not documented
9:21
I know that .logicdata viewer is amazing
9:21
I made two assumptions: that analog files would also be Blot documented and viewer would be amazing

SVeilleux9, 2021-03-01:
What data are you looking for from the Saleae? Or can you explain what end goal you are looking for?

closed 2025-06-10 by rusefillc