rusefi/rusefi #2440: bench test should be more precise

rusefillc, 2021-03-10:
at the moment (fuel and else) bench test commands accept fractional input but in the end only use integer part of input

```
		chThdSleepMicroseconds(onTimeUs);
		output->setValue(false);
		chThdSleepMicroseconds(offTimeUs);

```

JPh is asking for this to be more precise and use the fractional number not just intereger

mck1117, 2021-03-10:
We're limited on the accuracy of `chThdSleep*` because we tick at 1ms - so we can only sleep for integer multiples of 1ms.

mck1117, 2021-03-10:
old system was pretty terrible - other threads could preempt, so we were really only accurate within a few hundred microseconds, which is pretty useless for injectors that have opening times of ~300us these days.

PR #2441 (2021-03-10): accurate bench test
fix #2440
closed 2021-03-10 by rusefillc