rusefi/rusefi #2451: gen_config.sh code duplication

rusefillc, 2021-03-13:
At the moment we have one universal and three custom gen_config.sh scripts

we have "-firing_order controllers/algo/firing_order.h" and many related magic copy-pasted four times

we need to reduce copy-paste between those four files

chuckwagoncomputing, 2021-05-09:
The custom scripts have:
```
 -with_c_defines false \
 -initialize_to_zero false \
 -c_defines ${BOARD_DIR}/config/controllers/algo/rusefi_generated.h \
 -c_destination ${BOARD_DIR}/config/controllers/algo/engine_configuration_generated_structures.h \
```
Additionally, the subaru_eg33 script has:
```
 -enumInputFile ${BOARD_DIR}/rusefi_hw_enums.h \
```

What do the first two lines do?
What is the purpose of the second two? Why the custom output locations?
Why does the subaru board need it's own rusefi_hw_enums.h?

rusefillc, 2021-05-09:
Some poor documentation pushed about two lines

custom hw enum is related to

```
	MC33810_0_OUT_0 = 184,
	MC33810_0_OUT_1 = 185,
	MC33810_0_OUT_2 = 186,
	MC33810_0_OUT_3 = 187,
	MC33810_0_GD_0  = 188,
	MC33810_0_GD_1  = 189,
	MC33810_0_GD_2  = 190,
	MC33810_0_GD_3  = 191,

	MC33810_1_OUT_0 = 192,
	MC33810_1_OUT_1 = 193,
	MC33810_1_OUT_2 = 194,
	MC33810_1_OUT_3 = 195,
	MC33810_1_GD_0  = 196,
	MC33810_1_GD_1  = 197,
	MC33810_1_GD_2  = 198,
	MC33810_1_GD_3  = 199,
```

i.e. we have pretty questionable considerations for external GPIO handling. Some boards have one external TLE8888 port extended while others have two MC33810 and all this has to live in a byte for now.

chuckwagoncomputing, 2021-05-10:
also there is ` -Drusefi.generator.lazyfile.enabled=true \`
What does this do?

rusefillc, 2021-05-12:
@chuckwagoncomputing ``rusefi.generator.lazyfile.enabled`` was dead code, it is no longer with us

commit db03f253 (2021-07-31): gen_config.sh code duplication fix #2451
closed 2021-07-31 by rusefillc