rusefi/rusefi #2454: We shall mount a virtual mass storage device with current .ini file

rusefillc, 2021-03-13:
it should be possible to have rusEFI mount a mass storage device USB drive with rusefi.ini

rusefillc, 2021-03-13:
@mck1117 says that for uncompressed .ini to be displayed we would need 32Kb of RAM for this feature. That's because internally the .ini has to be compressed (because why?) and for rusEFI to handle the uncompression transparently we would need RAM.

looks like uncompressed feature would only work on 427+ chips, I would suggest runtime check "is 407 or 427"

mck1117, 2021-03-13:
The current implementation works by creating a VFAT (aka fat16 + long filename support) filesystem at build time that contains the ini, and storing that filesystem in flash on the ECU.  The ECU then presents that image to the PC as if it were a real drive, so the PC can read the data stored within as a native block device.

There are 4 conceivable options.

1. Put the ini file in the filesystem, and store the filesystem on the ECU.  No compression.
2. Compress the ini in to `rusefi.ini.zip`, then store that in the filesystem.  (compressed file, uncompressed filesystem).
3. Put the ini file in the filesystem, compress the whole filesystem, then store that in the ECU.
4. Compress the ini in to `rusefi.ini.zip`, store in the filesystem, then compress the filesystem, then store that in the ECU.

First, we can cross off option 4.  Compressing twice will not only only increase the size slightly but still add the complexity.

Next, we can cross off option 1. The plain .ini file is 750-850KB.  With a microcontroller with only 1MB of available flash, and ~350KB current code size, plus two 128K flash pages for config storage, that adds up to ~1.5MB total.  So SOME compression is required.  Thankfully the 800K ini file compressed down to <75K since it doesn't actually contain very much entropy.

The easiest option is number 2: The ECU can be completely dumb, and just offer up bytes, letting the PC decompress the zip file.  However - this presents a usability problem: you can't open `rusefi.ini.zip` directly from TunerStudio.

The "best experience" option is number 3: As far as the user can tell, it's just a plain ini file that you can open directly from TunerStudio.  However, since decompression of the entire filesystem image has to happen on the fly inside the ECU, it requires an extra 32K of RAM to store the DEFLATE decompression dictionary.

Options 2 and 3 have a similar flash storage footprint, as the compression of the ini alone vs. whole filesystem yields similar results (99% of the content is the same either way, the filesystem is effectively just a "header" on the front of the zip file).

mck1117, 2021-03-13:
related to #2417 and #2420

rusefillc, 2021-05-13:
@mck1117 is this done or not done?

mck1117, 2021-05-13:
looks done to me!

closed 2021-05-13 by mck1117