rusefi/rusefi #2457: VAG MQB crc support

rusefillc, 2021-03-14:
Would we need VAG CRC even for tach packet?

https://github.com/commaai/opendbc/blob/962c0c18e8b6638e62d786d6bebb08846b177ef4/can/common.cc#L95

mck1117, 2021-03-14:
wait, they add an additional CRC in to the payload? CAN already has a CRC built in to the protocol...

rusefillc, 2021-03-14:
it seems to have constant salt for less important packets and a cycle of salt for more safety related. seems a bir if a watchdoggy thing

mck1117, 2021-03-14:
That's not my question - is the CRC in the **DATA** field, or is it the **normal CAN bus** CRC?

Either way we might need extra data in the data fields.

rusefillc, 2021-03-15:
https://github.com/commaai/opendbc/blob/962c0c18e8b6638e62d786d6bebb08846b177ef4/can/common.cc#L102 hints at crc in data?