rusefi/rusefi #2462: MRE adapter 94 documentation

rusefillc, 2021-03-15:
I think i found the issue with the no comms while in the car. Pin 1S is for usb data- but it goes to the heater controller on the chassis side.

I have removed jp23 and 24 to isolate it from the chassis

stefanst, 2021-03-15:
Correct. Pin 1S had caused trouble before.

rusefillc, 2021-03-15:
@stefanst please make notes on relevant wiki pages

mi-hol, 2023-01-26:
@rusefillc I'd be happy to update wiki but due to missing URL I'm unclear and search has no good hits but only pdf docs.
What is the wiki URL?
[image]

mi-hol, 2023-01-26:
> don't care

is a strange concept to me because somebody will read the wrong information.
Only option I see is delete or correct!

rusefillc, 2023-01-27:
Delete

mi-hol, 2023-01-27:
@rusefillc happy to delete but need required permissions granted for organization rusefi (see my slack message)

mi-hol, 2023-01-27:
can below MRE adapter docs be deleted too?
[image]

closed 2023-01-28 by rusefillc