rusefi/rusefi #2470: Engine will not crank

biggie254-1, 2021-03-17:
Got a base file for a 190e M111 conversion. Have verified the wiring is ok with injector and ignition tests working correctly. Signal from camshaft sensor is received to the ECU but the injectors and coil packs will not fire on ignition. 

Could the error below be preventing the engine from cranking?

[image]

rusefillc, 2021-03-17:
rusEFI does not have a base tune for M111 - we have a legacy tune which has worked for some legacy version of the software.

Latest software requires both ETB sensors to be configured, not an issue.

rusefillc, 2021-03-17:
@OrchardPerformance @mck1117 close as "won't fix"?

mck1117, 2021-03-17:
Closing as "not a bug". This is the intended behavior. On vehicles with electronic throttle, we require redundant sensors to be configured.

In turn, any fatal config error will inhibit starting the engine.

closed 2021-03-17 by mck1117