rusefi/rusefi #2517: MAP decoding info not included into "tempinfo"

rusefillc, 2021-04-01:
tempinfo does not print MAP decoding

```
2021-04-01_17_14_05_784: EngineState: Sensor "CLT": Raw value: 1.65 Valid: Yes Converted value 35.21
2021-04-01_17_14_05_784: EngineState:     1.65 volts -> 1330.3 ohms with supply voltage 5.00 and pullup 2700.0.
2021-04-01_17_14_05_784: EngineState:     1330.3 ohms -> valid: Yes. 35.2 deg C
2021-04-01_17_14_05_784: EngineState: Sensor "IAT": Raw value: 1.97 Valid: Yes Converted value 28.21
2021-04-01_17_14_05_784: EngineState:     1.97 volts -> 1757.1 ohms with supply voltage 5.00 and pullup 2700.0.
2021-04-01_17_14_05_784: EngineState:     1757.1 ohms -> valid: Yes. 28.2 deg C
2021-04-01_17_14_05_784: EngineState: RPM sensor: stopped: 1 spinning up: 0 cranking: 0 running: 0 rpm: 0.000000000
2021-04-01_17_14_05_784: EngineState: Sensor "MAF" is not configured.
2021-04-01_17_14_05_784: EngineState: Sensor "Oil Pressure" is not configured.
2021-04-01_17_14_05_794: EngineState: Sensor "Fuel Pressure (LP)" is not configured.
2021-04-01_17_14_05_794: EngineState: Sensor "Fuel Pressure (HP)" is not configured.
2021-04-01_17_14_05_794: EngineState: Sensor "Fuel Pressure (injector)" proxied from sensor "Fuel Pressure (LP)"
2021-04-01_17_14_05_794: EngineState: Sensor "TPS 1" is redundant combining "TPS 1 Primary" and "TPS 1 Secondary"
2021-04-01_17_14_05_794: EngineState: Sensor "TPS 1 Primary": Raw value: 0.57 Valid: Yes Converted value 0.17
2021-04-01_17_14_05_794: EngineState:     Linear function slope: 31.79 offset: -17.96 min: -10.0 max: 110.0
2021-04-01_17_14_05_794: EngineState:       raw value 0.57 converts to 0.17 valid: Yes
2021-04-01_17_14_05_794: EngineState: Sensor "TPS 1 Secondary" is not configured.
2021-04-01_17_14_05_794: EngineState: Sensor "TPS 2" is redundant combining "TPS 2 Primary" and "TPS 2 Secondary"
2021-04-01_17_14_05_794: EngineState: Sensor "TPS 2 Primary" is not configured.
2021-04-01_17_14_05_794: EngineState: Sensor "TPS 2 Secondary" is not configured.
2021-04-01_17_14_05_794: EngineState: Sensor "Acc Pedal" is redundant combining "Acc Pedal Primary" and "Acc Pedal Secondary"
2021-04-01_17_14_05_794: EngineState: Sensor "Acc Pedal Primary" is not configured.
2021-04-01_17_14_05_794: EngineState: Sensor "Acc Pedal Secondary" is not configured.
2021-04-01_17_14_05_794: EngineState: Sensor "Driver Acc Intent" proxied from sensor "TPS 1"
2021-04-01_17_14_05_794: EngineState: Sensor "Aux Temp 1" is not configured.
2021-04-01_17_14_05_794: EngineState: Sensor "Aux Temp 2" is not configured.
2021-04-01_17_14_05_794: EngineState: Sensor "Lambda 2" is not configured.
2021-04-01_17_14_05_794: EngineState: Sensor "Wastegate Position" is not configured.
2021-04-01_17_14_05_804: EngineState: Sensor "Idle Valve Position" is not configured.
2021-04-01_17_14_05_804: EngineState: Sensor "Flex Fuel" is not configured.
2021-04-01_17_14_05_804: EngineState: Sensor "Battery Voltage": Raw value: 2.11 Valid: Yes Converted value 12.39
2021-04-01_17_14_05_804: EngineState:     Linear function slope: 5.85 offset: 0.00 min: 0.0 max: 50.0
2021-04-01_17_14_05_804: EngineState:       raw value 2.11 converts to 12.39 valid: Yes
2021-04-01_17_14_05_804: EngineState: Sensor "Barometric Pressure" is not configured.
2021-04-01_17_14_05_804: EngineState: fan=No @ PD12
2021-04-01_17_14_05_804: EngineState: A/C relay=No @ NONE
2021-04-01_17_14_05_804: EngineState: confirmation_tempinfo:8
```

while
analoginfo
```
2021-04-01_17_14_38_234: EngineState: analogInputDividerCoefficient: 2.00
2021-04-01_17_14_38_234: EngineState: WARNING: ADC: hip9011 input is not configured
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for hip9011
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for fuel gauge
2021-04-01_17_14_38_244: EngineState: TPS1 Primary ADC5 ADC_SLOW PA4 adc=0.28/input=0.57v/divider=2.00
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for TPS1 Secondary
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for TPS2 Primary
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for TPS2 Secondary
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for LPF
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for HPF
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for pPS1
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for pPS2
2021-04-01_17_14_38_244: EngineState: CLT ADC13 ADC_SLOW PC2 adc=0.82/input=1.65v/divider=2.00
2021-04-01_17_14_38_244: EngineState: IAT ADC14 ADC_SLOW PC3 adc=0.98/input=1.97v/divider=2.00
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for AuxT1
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for AuxT2
2021-04-01_17_14_38_244: EngineState: MAF ADC11 ADC_SLOW PC0 adc=0.64/input=1.28v/divider=2.00
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for FSIO analog
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for FSIO analog
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for FSIO analog
2021-04-01_17_14_38_244: EngineState: ADC is not assigned for FSIO analog
2021-04-01_17_14_38_244: EngineState: AFR ADC2 ADC_SLOW PA1 adc=0.02/input=0.05v/divider=2.00
2021-04-01_17_14_38_244: EngineState: MAP ADC12 ADC_FAST PC1 adc=0.15/input=0.30v/divider=2.00
2021-04-01_17_14_38_254: EngineState: ADC is not assigned for BARO
2021-04-01_17_14_38_254: EngineState: ADC is not assigned for extKno
2021-04-01_17_14_38_254: EngineState: ADC is not assigned for OilP
2021-04-01_17_14_38_254: EngineState: ADC is not assigned for CJ UR
2021-04-01_17_14_38_254: EngineState: ADC is not assigned for CJ UA
2021-04-01_17_14_38_254: EngineState: ADC is not assigned for HIP9011
2021-04-01_17_14_38_254: EngineState: Vbatt ADC6 ADC_SLOW PA5 adc=2.11/input=12.38v/divider=5.85
2021-04-01_17_14_38_254: EngineState: confirmation_analoginfo:10
```

mck1117, 2021-05-31:
I would think that *temp*info would show *temp*erature information?

rusefillc, 2022-10-06:
> I would think that _temp_info would show _temp_erature information?

@mck1117 please clarify if you take on this question has changed since https://github.com/rusefi/rusefi/pull/1297/files#diff-8452e9516e1a2097b2272f2da63793b32f6429370b2b9dd029dd6ec61661c41b

mck1117, 2022-10-08:
your question was not "why does it show other info", but "why does it not show MAP info", which I don't understand.

let's try #4646

PR #4646 (2022-10-10): consolidate analog vs. temp info commands
fix #2517 by doing fewer confusing things
closed 2022-10-10 by rusefillc